zephyrproject-rtos/zephyr #61589: Recover NET_ROUTE and NET_ROUTING

Hiihtaja77, 2023-08-17:
https://github.com/zephyrproject-rtos/zephyr/blob/45dcc6c5dbfb98944133a7c046959d763021386a/subsys/net/ip/Kconfig#L312

Routing look as hidden capability in Zephyr.
Can it be as generic part of Zephyr OS and support at list 3 network interfaces.

Hiihtaja77, 2023-08-18:
Typical Iot device might have connectivity to Host and also 2 or more different radio drivers.
IPV6 packet can arrive from Host and should be routed to desired radio driver.
Packets can arrive from any of network interface and be routed to Host .

M1cha, 2024-12-02:
Am I right in that this is an issue for openthread? Even with a border router, my zephyr-based thread device does not get a GUA, and can't even communicate with other ULAs outside of it's own, because there is no default route.

If that's true, it means that zephyr devices currently can only talk within a thread network and can't really utilize border routers.

rlubos, 2024-12-02:
@M1cha Not really, these options should not affect OpenThread device capability to reach external networks. If the device doesn't get GUA it may suggest some misconfiguration, please check the following issue where we investigated similar case: https://github.com/zephyrproject-rtos/zephyr/issues/29567#issuecomment-717311081