zephyrproject-rtos/zephyr #61600: Zephyr stdio has broken newline handling

rszkodzinski, 2023-08-17:
Zephyr stdin hook eats newline characters. Zephyr stdout hook adds \r after each \n.
This seems to be applied for all libc implementations, but I have tested mostly with newlib.

Unfortunately that is not what libc implementations expect - they expect to remove newlines as they see fit and expect to write the bytes they actually want to write.
This makes scanf, printf, read, write etc. broken in some cases when one cannot actually receive or send newline characters correctly.

carlescufi, 2023-08-17:
@rszkodzinski thanks for the issue. Would you consider opening a Pull Request with the fix?

rszkodzinski, 2023-08-17:
Unfortunately I'm somewhat busy writing completely unrelated code and there are bound to be unintended consequences in console or shell code for example.
Fortunately for me the hooks are not broken for getchar case, only for all the reads longer than 1.

carlescufi, 2023-08-22:
@rszkodzinski the hooks are not installed by default AFAIK, see here:
https://github.com/zephyrproject-rtos/zephyr/issues/60741#issuecomment-1649886273

Did you modify the code to install hooks?

rszkodzinski, 2023-08-23:
I did install the input hooks, yes. In particular, one that does RTT waitkey. (I was using RTT for testing.)

The output has the hooks installed by default, and those have interesting newline handling in them too.

carlescufi, 2023-08-23:
> I did install the input hook, yes, in my main. In particular, one that does RTT waitkey. (I was using RTT for testing.)

Thanks for the clarification and code @rszkodzinski. I will follow up.

closed 2023-11-06 by github-actions[bot]