zephyrproject-rtos/zephyr #61606: Provide sample for GATT Dynamic DB utilization for creation of GATT Services on runtime

hmxmghl, 2023-08-17:
**Is your enhancement proposal related to a problem? Please describe.**
Not exactly a problem but there is a lack of documentation on the utilization of available APIs to create resources for dynamic creation of GATT services on runtime.

**Describe the solution you'd like**
A working sample where a service is being created at runtime (triggered by some action, like when connection callback is called). Sample should include creation of all resources needed by the service to be created and registered at runtime.

**Describe alternatives you've considered**
Been trying to do what I described above but unfortunately, I am not able to set the resources correctly for dynamic service creation.

**Additional context**
I think I have provided all the information above.

hermabe, 2023-08-17:
The tester application used for host qualification tests does this, see https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/bluetooth/tester/src/btp_gatt.c.

I would not recommend creating services this way, it is much simpler to just use the helper macros to define a static service.

hmxmghl, 2023-08-17:
Thanks for the quick response! I know creating services dynamically can create many issues. And as you mentioned, whenever GATT DB is changed, a rediscovery by client will also be needed. But I have a use-case that I wanted to validate with this approach.

What is the recommended way to at least modify the existing service (created statically) on runtime. For example, add a new characteristic to an existing service.

towen, 2023-10-20:
I also have a use-case for this:
I am making a simple multi-channel temperature sensor using ds18b20 devices. I would like to be able to: 
1. plug in a previously-unused ds18b20 device to the bus (they have unique 64bit serial numbers in ROM)
2. initiate a search via a client command or button press to discover the device (this is working via button press)
3. create a new temperature sensor GATT service to display its UUID and sensor value
4. If possible, allow the user to set the device's name (and store the name in flash/eeprom, rather than on the client)

Currently I have one channel of data being returned by copying the `peripheral_ht` sample, but that seems to support only one sensor.
I was rather horrified by the complexity of the `BT_GATT_SERVICE_DEFINE` macro in the `peripheral_esp` sample.

hermabe, 2023-10-20:
> I also have a use-case for this: I am making a simple multi-channel temperature sensor using ds18b20 devices. I would like to be able to:
>
> 1. plug in a previously-unused ds18b20 device to the bus [...]

Does the client need to know/care about the serial number? Can multiple sensors be connected at the same time? You could potentially use a control point characteristic to select the sensor instead of adding a separate characteristic for each sensor.

Potential solution:
- Have one characteristic to read the temperature
- Have one characteristic to select which sensor to read from
- Have a characteristic to read which sensors are available? Or the server can notify the client when a new one is connected.

My idea here is that when adding a new sensor you change the _values_ in the database, not the _layout_.

Thalley, 2024-07-03:
Removed low priority bug label