zephyrproject-rtos/zephyr #61629: driver: ramdisk: add user allocation of ramdisk buffer

kunoh, 2023-08-18:
**Is your enhancement proposal related to a problem? Please describe.**
I think it would be useful if the ramdisk could be used as a general purpose method of sending arbitrary data to a device via USB. The use-case is a imxrt1060 mcu with LCD display, where we use the ramdisk for 2 framebuffers shared with the display_mcux_elcdif driver. This way the "dd" linux command can be used to write directly to the display framebuffer, without any extra copying needed to update the display.

**Describe the solution you'd like**
I propose to add configuration option CONFIG_DISK_RAM_USER_ALLOCATION to the ramdisk driver:
 
```Kconfig
config DISK_RAM_USER_ALLOCATION
    bool "RAM buffer is allocated by User"
    help
      RAM buffer is allocated in user space.
      Address of RAM buffer is set by ioctl command DISK_IOCTL_SET_RAMBUF_ADDR
```
 

With this configuration set, the ramdisk driver does not allocate the rambuffer, instead the buffer is allocated by the user and the ioctl disk command DISK_IOCTL_SET_RAMBUF_ADDR, is used to configure the ramdisk buffer address eg.:

```c
uint32_t cmd_buf = (uint32_t)p_buf;
if (disk_access_ioctl(disk_pdrv, DISK_IOCTL_SET_RAMBUF_ADDR, &cmd_buf) != 0) {
    LOG_ERR("Disk ioctl DISK_IOCTL_SET_RAMBUF_ADDR failed");
}
```

Laczen, 2023-08-21:
@kunoh I think the functionality you want is already supported by using flashdisk instead of ramdisk and the flash simulator. The flash simulator allows you to link a memory region to the flash simulator. The flash simulator and flashdisk can be setup to support 512kB block sizes instead of the standard 4196 kB flash block sizes so that there is no "mismatch" or flashdisk cache needed.

kunoh, 2023-08-22:
Thank you @Laczen, seems like the flash simulator could be used.

Do you know about the efficiency of using the flash simulator compared to the proposed changed to ramdisk. Ramdisk is more a more "barebone" solution since it only copies the data directly to ram. 
The flash simulator seems to be simulating flash disk operation suck as erase which may require more time, and as the intention for my use case is to use it as a framebuffer, I would like the highest speed possible to maximamize my framerate.

jfischer-no, 2023-08-22:
> ramdisk should allow you the double framerate compared to using it over flashdisk (flashdisk will erase, which is a memset) before doing a write. There is a proposed PR (#60149) on which I proposed to allow pointing it to a memory region. This would also solve your problem I think.

I will add an optional property for this.

kunoh, 2023-08-23:
Cool, thank you !

closed 2023-09-18 by carlescufi