zephyrproject-rtos/zephyr #61631: Low Power Mode backup timer

semihalf-niedzwiecki-dawid, 2023-08-18:
## Introduction

Add an additional IDLE timer actions, for system timers that are not clocked in a low power mode.

### Problem description

Some system timers are not clocked in a low power mode (LPM). The core example is SysTick in STM32 chips (not sure how other chips works).  Some chips use low power timer that runs in the LPM e.g. STM32Lx use LPTIM, but some chips families don't have such feature, e.g. STM32Fx. That means chips that use the SysTick as the system timer can't enter the low power mode. See #60558 

### Proposed change

Add a "backup/lpm" timer that is used while entering the low power mode. The additional timer would be used to wake up a chip and measure time missed by the main timer.

## Detailed RFC

Add a chosen node e.g. `zephyr,idle-timer`. If defined, the timer would be used in the `sys_clock_set_timeout` function to set an alarm to wake up the chip and save current time. The function takes `idle` as an argument, which indicates that we are going to enter low power mode. 

Then, in the `sys_clock_idle_exit` function, we should be able to calculate time missed by the main timer, update the cycle counter and announce it to the kernel. 

My assumption in that the `zephyr,idle-timer` node should support the counter API.

For the mentioned STM32Fx chips, the only timer/counter that runs in LPM is RTC, so it would be the idle timer. The maximum frequency that can be used for such case is 16kHz (LSI/2), so really not accurate comparing to the main clock.

### Concerns and Unresolved Questions

- How to announce the time to the kernel - call the `sys_clock_announce` in `sys_clock_idle_exit` or update the cycle counter and call `sys_clock_set_timeout(0, false)`
- It increases the exit latency/min residency, because of additional operation, but the alternative is no low power mode at all so it still sounds like a good deal
- Some inaccuracies in time measurement are unavoidable, due to difference in the clock frequencies 

## Alternatives

Use the idle timer as the main system timer. The RTC with 16kHz means 62,5us clock cycle, which is really low.

avolkov-1221, 2023-08-18:
Hi Dawid, 

It looks like our tickets are related (my is #61618). 

IMHO It will be better to use counter RTC as a system tick source (default system tick is 10KHz or 100Hz) and SYSTICK timer  as a cycle reference for sys_counter_cycle_get_XX (in this case it can be stopped in the idle). 

I suspect that your suggestion to recalibrate the system timer after every idle would be a nightmare.

semihalf-niedzwiecki-dawid, 2023-08-21:
Hello Andrey,
thanks for the response.

I've been considering both options here #60558. I've decided to start with the presented solution because of the low frequency of a RTC clock. For STM32 it would be 16kHz. The frequency should be high enough for the 100Hz system tick, but for system with higher requirement it may not be enough.

Regarding #61618, separation system timer driver and cycle counter sounds like a nice option, but I'm not familiar enough how the cycle counter is used to give you more feedback. Just please keep in mind, that some clocks might be disabled in low power mode and some can still run. If system timer and cycle counter use different clock the measurements may be shifted because of that e.g. system timer RTC uses LSE (running in LPM) and clock counter uses PLL that is disabled while entering LPM.

>I suspect that your suggestion to recalibrate the system timer after every idle would be a nightmare.
I'm not sure what you mean by `recalibrate`. The main system timer would run without any changes and the idle timer would measure the "missed" time and set alarm to wake up the device.

avolkov-1221, 2023-08-23:
> Hello Andrey, thanks for the response.
>
> I've been considering both options here #60558. I've decided to start with the presented solution because of the low frequency of a RTC clock. For STM32 it [...]

But tick frequency could also be power of 2. 100 HZ/10 KHz is just anthropocentric numbers. Why not use a 16384/8192 Hz system timer instead, if it's acceptable for your application?

> 
> Regarding #61618, separation system timer driver and cycle counter sounds like a nice option, but I'm not familiar enough how the cycle counter is used to give you more feedback. Just please keep in mind, that some clocks might be disabled in low power mode and some can still run.

Yes, I know, this STM's (and Renesas) decision has already made be headache.

> If system timer and cycle counter use different clock the measurements may be shifted because of that e.g. system timer RTC uses LSE (running in LPM) and clock counter uses PLL that is disabled [...]

You will need to keep 2 timers (backup and system) in sync, and it will not be so easy since LPM timer is 32,768 KHz based, but high precision system one is 64 MHz (or more) based. So after a couple of days of running the chip, you will have a leap error, if you do not take some action. The "missing" time is a fractional value in therms of system/or backup timer.

semihalf-niedzwiecki-dawid, 2023-08-25:
>But tick frequency could also be power of 2. 100 HZ/10 KHz is just anthropocentric numbers. Why not use a 16384/8192 Hz system timer instead, if it's acceptable for your application?

Yes yes, I know. The max freq for STM32 RTC is 16kHz. However, we are migrating our system with much higher frequency to Zephyr. I want to keep all parameters as much as possible. It the proposed solution doesn't work, I will try with the RTC as system timer.

>You will need to keep 2 timers (backup and system) in sync,

That is not actually true. Every time before entering the LPM and after exiting, I would read the time from both timers. I allows calculating the passed time. The difference between measurements is the time missed by the main timer, and cycle counter should be increased with this value.
The process would be repeated for every idle entrance, the backup timer doesn't keep its version of the cycle counter.

niedzwiecki-dawid, 2023-11-24:
Added with #63187

closed 2023-11-24 by niedzwiecki-dawid