zephyrproject-rtos/zephyr #61644: Possible issue within nvs_sector_close

Valle125, 2023-08-18:
Hello,

there is an unhandled return value within nvs_sector_close (subsys/fs/nvs/nvs.c line 558). 
For me it's not clear weather the return value is discarded intentionally or if it should actually be returned to the 
caller of the function.
If I compile the code with -Wall enabled I get a -Wunused-but-set-variable  warning at line 542. That's how I discovered this issue.

**Expected behavior**
I'd suggest to either return rc to the caller, e.g.
`rc = nvs_flash_ate_wrt(fs, &close_ate);
if (rc)
  return rc;
`
or casting the return value to void, e.g.
`(void)nvs_flash_ate_wrt(fs, &close_ate);`

**Impact**
This has low priority and no impact to my project.

Best,
Valentin

fabiobaltieri, 2023-08-22:
Hi, Zephyr is an open-source project, and as such we welcome and encourage the community to submit patches directly. Would you be able to submit a pull request with the fix? This is the [contribution guideline documentation](https://docs.zephyrproject.org/latest/contribute/guidelines.html#contribution-guidelines).

Valle125, 2023-08-23:
I can submit a patch tomorrow.

PR #61646 (2023-09-02): nvs: remove unused return code in nvs_sector_close
Fixes #61644
commit cb19fc7a (2023-09-02): nvs: remove unused return code in nvs_sector_close

Fixes #61644
closed 2023-09-02 by carlescufi