zephyrproject-rtos/zephyr #61647: Promiscuous mode altering some packets

graffiti, 2023-08-18:
I am using Zephyr for an ethernet application on an STM32H7. I use promiscuous mode on the interface and net_promisc_mode_wait_data() to successfully receive all ethernet packets, however, some packets have their ethernet headers stripped, and others don't.

In net_if.c net_if_recv_data() is the following code:

```C
verdict = net_if_l2(iface)->recv(iface, pkt);
if (verdict == NET_CONTINUE) {
new_pkt = net_pkt_clone(pkt, K_NO_WAIT);
} else {
new_pkt = net_pkt_ref(pkt);
}
```

This passes the packet to the layer 2 handler BEFORE it's cloned and passed to the promiscuous fifo. If the layer 2 hander (ethernet.c ethernet_recv() ) doesn't care about the packet, then verdict is NET_DROP and the packet gets placed in the promiscuous fifo unmodified. If however, the layer 2 hander also wants to handle the packet then it strips the headers by advancing the buffer pointer (net_buf_pull()), and verdict is NET_CONTINUE. It is only AFTER this that the packet is cloned and added to the promiscuous fifo.

Thus, some packets (multicast, broadcast, non self target mac address packets) are received through the promiscuous API unmodified, while others (those targeted to the MAC address of the device) are received with the ethernet headers stripped.

Suggestion: always clone the packet and pass to the promiscuous fifo before passing the original pkt up the stack, by replacing the following code in net_if.c net_if_recv_data():

```C
if (IS_ENABLED(CONFIG_NET_PROMISCUOUS_MODE) &&
	    net_if_is_promisc(iface)) {
		/* If the packet is not for us and the promiscuous
		 * mode is enabled, then increase the ref count so
		 * that net_core.c:processing_data() will not free it.
		 * The promiscuous mode handler must free the packet
		 * after it has finished working with it.
		 *
		 * If packet is for us, then NET_CONTINUE is returned.
		 * In this case we must clone the packet, as the packet
		 * could be manipulated by other part of the stack.
		 */
		enum net_verdict verdict;
		struct net_pkt *new_pkt;

		/* This protects pkt so that it will not be freed by L2 recv()
		 */
		net_pkt_ref(pkt);

		verdict = net_if_l2(iface)->recv(iface, pkt);
		if (verdict == NET_CONTINUE) {
			new_pkt = net_pkt_clone(pkt, K_NO_WAIT);
		} else {
			new_pkt = net_pkt_ref(pkt);
		}

		/* L2 has modified the buffer starting point, it is easier
		 * to re-initialize the cursor rather than updating it.
		 */
		net_pkt_cursor_init(new_pkt);

		if (net_promisc_mode_input(new_pkt) == NET_DROP) {
			net_pkt_unref(new_pkt);
		}

		net_pkt_unref(pkt);

		return verdict;
	}
```

with:
```C
if (IS_ENABLED(CONFIG_NET_PROMISCUOUS_MODE) &&
	    net_if_is_promisc(iface)) {
		struct net_pkt *new_pkt;

		new_pkt = net_pkt_clone(pkt, K_NO_WAIT);

		if (net_promisc_mode_input(new_pkt) == NET_DROP) {
			net_pkt_unref(new_pkt);
		}
	}
```

Please let me know if you would like a PR.

Thanks, Kieran.

fabiobaltieri, 2023-08-22:
Hi, Zephyr is an open-source project, and as such we welcome and encourage the community to submit patches directly. Would you be able to submit a pull request with the fix? This is the [contribution guideline documentation](https://docs.zephyrproject.org/latest/contribute/guidelines.html#contribution-guidelines).

rlubos, 2023-08-29:
> Please let me know if you would like a PR.

@graffiti The proposed solution to clone the packet unconditionally looks correct, are you willing to open a PR with that fix?

graffiti, 2023-08-29:
>
>
> Please let me know if you would like a PR.
>
> @graffiti <https://github.com/graffiti> The proposed solution to clone
> the packet unconditionally looks correct, are you willing to open a PR with
> that fix?
>
Thanks for reviewing the proposal. Yes, I do plan to open a PR, with some
luck this week...

Thanks, Kieran.

Message ID: ***@***.***>
>

PR #63408 (2023-10-03): net: ip: fix promiscuous mode altering packets
Always clone net_pkt to pass to promiscuous queue. #61647
Previously, net_pkt was passed to L2 before conditionally cloning. But L2 would in some cases strip ethernet headers, so cloned net_pkt received through promiscuous interface would be missing headers.

Fixes #61647
closed 2023-10-03 by fabiobaltieri