zephyrproject-rtos/zephyr #61654: Nordic CONFIG_GPIO_AS_PINRESET not working

Fattoresaimon, 2023-08-18:
I'm working on a nRF52805 with Visual Studio code.
Zephyr version: 3.2.99 build: v3.2.99-ncs2

I need to use the RESET pin as a GPIO and looks like the configuration is not working.
Writing **CONFIG_GPIO_AS_PINRESET=n** or **CONFIG_GPIO_AS_PINRESET=y** it doesn't change the result.
The PSELRESET register in the UICR  is always **:15000000** that means RESET on GPIO 21
The only way to change the configuration is to erase the UICR register with **nrfjprog --eraseuicr** after every  programming.

Thank you!
Simone

anangl, 2023-08-21:
> The only way to change the configuration is to erase the UICR register with **nrfjprog --eraseuicr** after every programming.

If you use an application with `CONFIG_GPIO_AS_PINRESET=n`, you need to erase UICR before flashing it if the nRESET functionality was enabled earlier. It it was not enabled, there is no need to erase UICR.
When you flash an application with `CONFIG_GPIO_AS_PINRESET=y` (which is the default setting), UICR will be modified and the SoC will be reset automatically if needed. Take a look [here](https://github.com/zephyrproject-rtos/hal_nordic/blob/9ae7c765985ebdea3d9b98c0d3b154794f0b47cf/nrfx/mdk/system_nrf52.c#L339-L350) for the relevant part of `SystemInit`.

Fattoresaimon, 2023-08-22:
Hello, thank you for the quick answer.
I have tried your suggestion using Visual studio

- Set `CONFIG_GPIO_AS_PINRESET=n` in the .prj file
- erase UICR
- flash again
- read back the UICR

But i still get:
`:101200001500000015000000FFFFFFFFFFFFFFFFBC`
means the GPIO is set as a RESET, not as a GPIO
Checking in the function` void SystemInit(void)` 
i see the part of `#if defined(CONFIG_GPIO_AS_PINRESET)` is commented out
so the code is working but some how the register is still set when i flash the device.

Now i have made another test:
If i flash the .hex file with J-Flash and not with Visual Studio, it works in both cases:
With  `CONFIG_GPIO_AS_PINRESET=n` i see the UICR register all 0xFF
And with  `CONFIG_GPIO_AS_PINRESET=y` i see `:101200001500000015000000FFFFFFFFFFFFFFFFBC`

So looks like the issue is in Visual Studio when it flash the device

carlescufi, 2023-08-23:
Closing since this seems to be related to VSCode and not to Zephyr itself.

closed 2023-08-23 by carlescufi