zephyrproject-rtos/zephyr #61657: native_posix: custom applications build has been broken since missing 'pinctrl_soc.h'

avolkov-1221, 2023-08-18:
**Describe the bug**
The introduction of obligatory board specific "pinctrl_soc.h" file has broke all the **native_posix** custom application builds.

**To Reproduce**
Just try to build any real hardware related application for the native_posix/native_posix_64 target. The build will be crashed due to 'missing pinctrl_soc.h'

aescolar, 2023-08-19:
Can you please provide an example of a sample in which this issue can be reproduced and describe what you expected would happen?

> The introduction of obligatory board specific "pinctrl_soc.h"

Can you please link to which commit/PR introduced that requirement.

avolkov-1221, 2023-08-19:
Sorry, I was too concentrate on the pinctrl problem, definitely the file is not obligatory for **"all"** but only for **"the pinctrl related"**.  The "pinctrl_soc.h" is a part of "include/drivers/pinctrl.h" since 3.0 at least,  so if the arch has not it, then app's build will crashed. 
We, as an example, from time to time are using posix target for our app internal, hardware independent, logic checking.

gmarull, 2023-08-21:
This is not an issue, `#include <zephir/driver/pinctrl.h>` can't be included in native_posix builds because there's no pinctrl in POSIX.

avolkov-1221, 2023-08-21:
> This is not an issue, `#include <zephir/driver/pinctrl.h>` can't be included in native_posix builds because there's no pinctrl in POSIX.

Could you please point to the doco where it's explained ? That API level headers cannot be included. 
Also could you explain why the **"tests/drivers/pinctrl/api"** test still exists ? I have to pay attention to the last part of the path.

gmarull, 2023-08-21:
Closing, as this is not describing any existing problem. pinctrl API test contains a pinctrl_soc.h designed for the test, and pinctrl API is not available on POSIX (there's no such thing in the POSIX environment).

closed 2023-08-21 by gmarull