zephyrproject-rtos/zephyr #61661: Incorrect docstring in atomic.h

hershjos, 2023-08-18:
zephyr/include/zephyr/sys/atomic.h: 165 is incorrect.

"* Atomically set bit number @a bit of @a target and return its old value.
 * The target may be a single atomic variable or an array of them."

The function returns the current value of the bit being set, not the old value. May seem minor but I've already dealt with supplier code that is incorrect because it assumes its behavior.

`https://github.com/zephyrproject-rtos/zephyr/blob/45a4177704247e22f45a699d3a376b8b564b4b7e/include/zephyr/sys/atomic.h#L165
`
Zephyr. Branch main, version 3-4-0.  Latest commit for the file is 24d09d363c59ca3d18dd24d3fc7c80582af56075

nashif, 2023-08-18:
> The function returns the current value of the bit being set, not the old value.

not sure I am following. This function is returning a boolean, not the current value...

fabiobaltieri, 2023-08-22:
Hi, Zephyr is an open-source project, and as such we welcome and encourage the community to submit patches directly. Would you be able to submit a pull request with the fix? This is the [contribution guideline documentation](https://docs.zephyrproject.org/latest/contribute/guidelines.html#contribution-guidelines).

nashif, 2023-08-22:
> not sure I am following. This function is returning a boolean, not the current value...

ok, this is an issue, i see the wrong language now.

kartben, 2023-08-22:
@hershjos Please let me know if you need help with the PR, I'll be happy to assist!

hershjos, 2023-08-22:
@nashif I realized the first half of the docstring is correct since atomic_or returns the previous and not current value. its the @return that's incorrect "* @return true if the bit was set, false if it wasn't."

I'm going on vacation but I'll improve the wording and submit a PR afterwards.

closed 2024-01-19 by github-actions[bot]