zephyrproject-rtos/zephyr #61664: I made a standalone menuconfig

JiauZhang, 2023-08-19:
Hey, guys.

I made a standalone menuconfig --> https://github.com/JiauZhang/menuconfig

Kconfig is an important configuration tool that is used in many projects. The commonly used Kconfig implementation, [Kconfiglib](https://github.com/ulfalizer/Kconfiglib), has not been updated since Jan 31 and 2020, so I created this project. Most of the source code is synchronized from the Linux kernel, and I have added some additional code to enable menuconfig to run independently.

I will regularly synchronize the relevant code in the Linux kernel into this project.

Welcome everyone to use it and provide good suggestions.

gmarull, 2023-08-21:
we use https://github.com/zephyrproject-rtos/zephyr/tree/main/scripts/kconfig

henrikbrixandersen, 2023-08-23:
@JiauZhang What are you suggesting?

closed 2023-08-23 by JiauZhang