zephyrproject-rtos/zephyr #61667: The snprintf / vsnprintf interface returns an error when the buffer size passed in is not enough to fill the characters

SeeDeer, 2023-08-19:
lib\libc\minimal\source\stdout\sprintf.c

int snprintf(char *ZRESTRICT str, size_t len,
	     const char *ZRESTRICT format, ...)
{
	va_list vargs;

	struct emitter p;
	int     r;
	char    dummy;

	if (len == 0) {
		str = &dummy; /* write final NUL to dummy, can't change *s */
	}

	p.ptr = str;
	p.len = (int) len;

	va_start(vargs, format);
	r = cbvprintf(sprintf_out, (void *) (&p), format, vargs);
	va_end(vargs);
       
        /* The following code should be added
	if (r > (int)len - 1) {
		ret = (int)len - 1;
	}  */

	*(p.ptr) = 0;
	return r;
}

keith-packard, 2023-08-22:
This relates to the C standard definition of snprintf and the implementation within the minimal C library. Here's a chunk of text from that standard

> Returns
> 
> The snprintf function returns the number of characters that would have been written
> had n been sufficiently large, not counting the terminating null character, or a negative
> value if an encoding error occurred. Thus, the null-terminated output has been
> completely written if and only if the returned value is nonnegative and less than n.

If snprintf returned the number of bytes _actually_ written rather than the number that _would_ have been written, the application would be unable to determine that the provided buffer wasn't large enough. I think the existing code is correct as written (even if it seems like an unexpected value to someone not familiar with the fine details of the C language spec).

fabiobaltieri, 2023-08-22:
Thanks for clarifying @keith-packard, closing down.

closed 2023-08-22 by fabiobaltieri