zephyrproject-rtos/zephyr #61669: Error in SD crad storage init

tanmaygoyal875, 2023-08-19:
I am using STM32F469i_disco which has inbuild SD-Card slot. I am using 16GB of sd-card. I have done every possible solution to resolve below error but cant please suggest me the solution for it. i am using the zephyr fs fat code for that and below the error i am putting dts file for that 

*** Booting Zephyr OS build zephyr-v3.4.0-1798-ga0a9539d0223 ***
[00:00:00.000,000] <err> main: Storage init ERROR!
[00:00:00.000,000] <err> fs: fs mount error (-19)
Error mounting disk.

~~~~~~~~~~~~~~~~~~~~~~~~~~~DTS~~~~~~~~~~~~
&flash0 {

	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;
		storage_partition: partition@d8000 {
			label = "storage";
			reg = <0x000d8000 0x00028000>;
		};
	};
};

/ {
	msc_disk0 {
		status="okay";
		compatible = "zephyr,flash-disk";
		partition = <&storage_partition>;
		disk-name = "SD";
		/* cache-size == page erase size */
		cache-size = <4096>;
	};
};
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Desvauxm-st, 2023-08-23:
@tanmaygoyal875
Can you tell me how to reproduce this problem.
do you use samples\subsys\fs\fs_sample for this?

tanmaygoyal875, 2023-08-23:
yes sir

Laczen, 2023-08-23:
@tanmaygoyal875 your setup is describing a disk on the internal flash of your device. This description is not honoring the flash erase-blocks of the device and this is why you are getting this error.

To enable the use of an SD-card slot your overlay file should use something similar to the one provided in https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/fs/fs_sample/boards/nucleo_f429zi.overlay

Desvauxm-st, 2023-08-24:
@tanmaygoyal875
I try  and i see the same error message  Storage init ERROR!
I will try to push a new PR to try to fix this bug 
regards

Desvauxm-st, 2023-08-24:
@tanmaygoyal875
can you try this modification:
boards: arm: stm32f469i_disco: add sdmmc1 #61859

closed 2023-10-02 by erwango