zephyrproject-rtos/zephyr #61681: NRF: pinctrl disregards PWM channel selected in DT and forces a default channel

marco85mars20, 2023-08-21:
**Describe the bug**
The current `pinctrl` implementation for NordicSemi microcontrollers disregards the channel selected in the device-tree (eg. via `pwms` property). For instance, on the nRF52840, there are 4 PWM peripherals and each of them provides 4 channels that can be used simultaneously, however `pinctrl` does not take into account the selected channel and for PWM`n` to use channel `n`, eg. PWM2 uses channel 2.

It all boils down to the implementation of API function `pinctrl_configure_pins()` in `zephyr/drivers/pinctrl/pinctrl_nrf.c`:
```
		case NRF_FUN_PWM_OUT0:
			NRF_PSEL_PWM(reg, OUT[0]) = pin;
			write = NRF_GET_INVERT(pins[i]);
			dir = NRF_GPIO_PIN_DIR_OUTPUT;
			input = NRF_GPIO_PIN_INPUT_DISCONNECT;
			break;
		case NRF_FUN_PWM_OUT1:
			NRF_PSEL_PWM(reg, OUT[1]) = pin;
			write = NRF_GET_INVERT(pins[i]);
			dir = NRF_GPIO_PIN_DIR_OUTPUT;
			input = NRF_GPIO_PIN_INPUT_DISCONNECT;
			break;
		case NRF_FUN_PWM_OUT2:
			NRF_PSEL_PWM(reg, OUT[2]) = pin;
			write = NRF_GET_INVERT(pins[i]);
			dir = NRF_GPIO_PIN_DIR_OUTPUT;
			input = NRF_GPIO_PIN_INPUT_DISCONNECT;
			break;
		case NRF_FUN_PWM_OUT3:
			NRF_PSEL_PWM(reg, OUT[3]) = pin;
			write = NRF_GET_INVERT(pins[i]);
			dir = NRF_GPIO_PIN_DIR_OUTPUT;
			input = NRF_GPIO_PIN_INPUT_DISCONNECT;
			break;
```

**To Reproduce**
Set a devicetree to use a PWM channel not matching the PWM peripheral ID.

**Expected behavior**

**Impact**
Without any workaround, NRF micros can only use up to 4 pwm-driven pins instead of 16, and care has to be put in selecting a channel matching the PWM ID.

**Logs and console output**
N/A.

**Environment (please complete the following information):**
Still present in the latest version in `main`

**Additional context**
N/A

anangl, 2023-08-21:
> It all boils down to the implementation of API function `pinctrl_configure_pins()` in `zephyr/drivers/pinctrl/pinctrl_nrf.c`:

Could you describe what exactly you think is wrong with this implementation?

> **To Reproduce**
> Set a devicetree to use a PWM channel not matching the PWM peripheral ID.

Like this, for example?
https://github.com/zephyrproject-rtos/zephyr/blob/5c901052ec2a105594a91e33b8fb43d40c34b88f/boards/arm/nrf52840dongle_nrf52840/nrf52840dongle_nrf52840.dts#L49-L60

marco85mars20, 2023-08-21:
I have been doing more investigation, and it appeared that the channel is retrieved from the pinctrl definition and not from `pwms`.
What is the purpose of having to specify the channel in the `pwms` property since now `pinctrl` is the way to go?

anangl, 2023-08-21:
> I have been doing more investigation, and it appeared that the channel is retrieved from the pinctrl definition and not from `pwms`, and I had selected mismatching channels. What is the purpose of having to specify the channel in the `pwms` property since now `pinctrl` is the way to go?

With `pinctrl` you assign a pin to a PWM channel, with `pwms` you select a given PWM channel to be used for a particular purpose.

closed 2023-08-21 by marco85mars20