zephyrproject-rtos/zephyr #61683: IO-Link Wireless implementation

IngFu, 2023-08-21:
IO-Link is an industrial communications networking standard (IEC 61131). IO-Link Wireless is a radio based extension to the standard, with a physical layer based on BLE 4.2
The protocol is specified here: https://io-link.com/share/Downloads/System-Extensions/IOLW-Spec_10112_V113_Jun22_5.pdf

Since IO-Link Wireless is becoming more and more widespread, an implementation as zephyr would be very useful.

henrikbrixandersen, 2023-08-23:
@IngFu Both IO-Link and IO-Link Wireless would be welcome additions. Do you plan on working on these yourself?

IngFu, 2023-08-25:
@henrikbrixandersen I am still in an early project phase. My company develops a wireless industrial sensor, that deploys BLE for connectivity. The option to also offer the  IO-Link Wireless protocoll using the same hardware and software would be very useful. However, I do not have the expertise to implement a radio protocoll and I am fairly new to zephyr. So right now I am investigating the feasibility and what kind of ressources would be necessary.