zephyrproject-rtos/zephyr #61692: Bluetooth: Controller: ISO: ASSERTION FAIL [c_latency <= cig->c_latency]

Thalley, 2023-08-21:
**Describe the bug**
There are 3 specific CAP tests that fails with `ASSERTION FAIL [c_latency <= cig->c_latency]`:

- Execute_AC_5 48_6_1 48_6_1 (single bidir CIS, 10ms, unframed, 310 octets C->P, 155 octets P->C, rtn 5, max latency 20ms)
- Execute_AC_11_I 48_5_1 48_5_1 (2 bidir CIS to a single peripheral, 7.5ms, unframed, 117 octets each way, rtn 5, max latency 15)
- Execute_AC_11_II 48_5_1 48_5_1 (1 bidir CIS to a two peripherals, 7.5ms, unframed, 117 octets each way, rtn 5, max latency 15)

**To Reproduce**
Steps to reproduce the behavior:
1. Build the LE Audio BSIM test (tests/bsim/bluetooth/audio)
2. Enable `Execute_AC_5 48_6_1 48_6_1` in `tests/bsim/bluetooth/audio/test_scripts/cap_unicast_ac_5.sh`, `Execute_AC_11_I 48_5_1 48_5_1` in `tests/bsim/bluetooth/audio/test_scripts/cap_unicast_ac_11_i.sh` and `Execute_AC_11_II 48_5_1 48_5_1` in `tests/bsim/bluetooth/audio/test_scripts/cap_unicast_ac_11_ii.sh`
3. Run any or all of the 3 above test scripts
4. See error

**Expected behavior**
Expect the tests to pass without an assertion

**Impact**
Makes it impossible to run certain configurations

**Logs and console output**
Failing part of log when running `Execute_AC_5 48_6_1 48_6_1` (the others are similar)

```
d_00: @00:00:05.935992  Source discover complete
d_00: @00:00:05.936097  stream 0x81ff540
d_00: @00:00:05.936097  named_preset 0x81ab1e4
d_00: @00:00:05.936097  stream 0x81ff940
d_00: @00:00:05.936097  named_preset 0x81ab1e4
d_00: @00:00:05.936097  [00:00:05.936,096] <dbg> bt_bap_unicast_client: bt_bap_unicast_client_new_audio_iso: New bap_iso 0x820f160
d_00: @00:00:05.936097  [00:00:05.936,096] <dbg> bt_bap_unicast_client: unicast_group_add_stream: group 0x8210700 stream 0x81ff940 qos 0x81ffd30 iso 0x820f160 dir 2
d_00: @00:00:05.936097  [00:00:05.936,096] <dbg> bt_bap_iso: bt_bap_iso_bind_stream: bap_iso 0x820f160 stream 0x81ff940 dir source
d_00: @00:00:05.936097  [00:00:05.936,096] <dbg> bt_bap_iso: bt_bap_iso_get_ep: iso 0x820f160 dir sink
d_00: @00:00:05.936097  [00:00:05.936,096] <dbg> bt_bap_unicast_client: unicast_group_add_stream: group 0x8210700 stream 0x81ff540 qos 0x81ff930 iso 0x820f160 dir 1
d_00: @00:00:05.936097  [00:00:05.936,096] <dbg> bt_bap_iso: bt_bap_iso_bind_stream: bap_iso 0x820f160 stream 0x81ff540 dir sink
d_00: @00:00:05.936097  [00:00:05.936,096] <dbg> bt_bap_iso: bt_bap_iso_get_ep: iso 0x820f160 dir source
d_00: @00:00:05.936097  [00:00:05.936,096] <dbg> bt_bap_unicast_client: bt_audio_cig_create: group 0x8210700 qos 0x81ffd30
d_00: @00:00:05.936097  ASSERTION FAIL [c_latency <= cig->c_latency] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_central_iso.c:504
d_00: @00:00:05.936097  @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_central_iso.c:504
d_00: @00:00:05.936097  [00:00:05.936,096] <err> os: z_fatal_error: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
d_00: @00:00:05.936097  [00:00:05.936,096] <err> os: z_fatal_error: Current thread: 0x8205e00 (BT TX)
d_00: @00:00:05.936097  [00:00:05.936,096] <err> os: k_sys_fatal_error_handler: Halting system
d_00: @00:00:05.936097 WARNING: Exiting due to fatal error
d_00: @00:00:05.936097  TESTCASE NOT PASSED at exit (test return (1) indicates it was still in progress)
```

**Environment (please complete the following information):**
 - Commit SHA or Version used: 6b654bdc870eaf85afa403d4957397823480bca2

**Additional context**
N/A

cvinayak, 2023-11-03:
@mtpr-ot I have been debugging this issue and noticed that, Controller is simply using the supplied RTN, and the calculated sync_delay used in calculating actual latency theoretically exceeds the max transport latency requested.  Does this mean, like in the BAP spec, Controller reduce the RTN down?  i.e. a RTN of 3 passes the test. 

Is this (Controller use reduced RTN) something we can/need to do to reduce the RTN in the controller to support the BAP 48_6_1 preset?

mtpr-ot, 2023-11-03:
> @mtpr-ot I have been debugging this issue and noticed that, Controller is simply using the supplied RTN, and the calculated sync_delay used in calculating actual latency theoretically exceeds the max transport latency requested. Does this mean, like in the BAP spec, Controller reduce the RTN down? i.e. a RTN of 3 passes the test.
> 
> Is this (Controller use reduced RTN) something we can/need to do to reduce the RTN in the controller to support the BAP 48_6_1 preset?
> 
> Failing: [image]
> 
> Passing: [image]

Yes, I would consider that a good policy to be be applied, to let max transport latency dictate reduction of retransmissions.

PR #64908 (2023-11-14): Bluetooth: Controller: Reduce RTN for requested Max Transport Latency
Add implementation to reduce CIG's CIS retransmissions so as to meet the Host requested Maximum Transport Latency.

Fixes #61692.
closed 2023-11-14 by carlescufi