zephyrproject-rtos/zephyr #61700: Bluetooth: Audio: bap_broadcast_assistant API inaccesible from recv_state callback

Chris-StJohn-Bose, 2023-08-21:
**Describe the bug**
BAP Broadcast Assistant uses a callback API to indicate remote state changes. An application would expect to use these callbacks to make API calls itself. However those API calls then fail with -EBUSY.

The problem is related to the use of Write Request in one direction and Notification in the other. The peer notification "overtakes" the Write Response so BAP Broadcast Assistant is unable to make further Write Requests.

**To Reproduce**
1. Zephyr is running at both ends (BAP Broadcast Assistant and BAP Scan Delegator)
2. Nordic nRF 5340 Audio DK boards - though the problem appears to be in high-level BAP support code so platform may be irrelevant
3. Boards Advertise/Scan, Connect, Bond
4. Client discovers Broadcast Audio Scan Service on Server
5. Client sends Add Source write request (BASS v1.0 3.1.1.4)
6. Server responds with Broadcast Receive State notification (BASS v1.0 3.2) containing Source ID
7. Client responds with Set Broadcast Code write request using Source ID (BASS v1.0 3.1.1.6)
8. `bt_bap_broadcast_assistant_set_broadcast_code()` fails -EBUSY

**Expected behavior**
`bt_bap_broadcast_assistant_set_broadcast_code()` API (and others...) should be available within `bt_bap_broadcast_assistant_cb::recv_state` callback

**Impact**
An application using the current behaviour would be required to duplicate much of the work of the BAP Broadcast Assistant by storing the Source ID from `recv_state` callback then waiting for `add_src` callback before progressing to Set Broadcast Code.

I modified the BAP Broadcast Assistant to use Write No Response rather than Write Request and this works fine. However this prevents most of the callbacks firing since they handle the Write Response which will no longer come. In order to proceed in this direction an architectural review would be required to establish whether these callback are actually useful.

**Logs and console output**
Note that these logs demonstrate the proposed solution of Write No Response, but they give an overview of the problem.

Server: 
```
HL [00:00:00.313,629] <inf> bt_hci_core: Identity: E9:95:8A:50:69:39 (random)
HL [00:00:00.313,659] <inf> bt_hci_core: HCI: version 5.2 (0x0b) revision 0x0d15, manufacturer 0x07e8
HL [00:00:00.313,690] <inf> bt_hci_core: LMP: version 5.2 (0x0b) subver 0x0d15
HL [00:00:00.314,544] <inf> ble: Controller version: 3349
HL [00:00:00.435,272] <inf> bis_headset: Advertising successfully started
HL [00:00:00.645,507] <inf> bis_headset: Connected: 14:D4:F9:A0:45:33 (public)
HL [00:00:01.060,546] <inf> ble: Passkey accepted for 14:D4:F9:A0:45:33 (public): 942432
HL [00:00:01.599,639] <inf> ble: Pairing completed: 14:D4:F9:A0:45:33 (public), bonded: 1
HL [00:00:04.839,752] <dbg> bt_bap_scan_delegator: recv_state_cfg_changed: value 0x0001
HL [00:00:05.019,744] <dbg> bt_bap_scan_delegator: read_recv_state: Index 0: Not active
HL [00:00:05.199,829] <dbg> bt_bap_scan_delegator: write_control_point: Data
                                                   02 00 00 00 00 00 00 00  00 56 34 12 00 ff ff 00 |........ .V4.....
HL [00:00:05.199,859] <dbg> bt_bap_scan_delegator: write_control_point: Assistant adding source
HL [00:00:05.199,859] <dbg> bt_bap_scan_delegator: scan_delegator_add_source: Index 0: New source added: ID 0x00
HL [00:00:05.200,012] <dbg> bt_bap_scan_delegator: bt_debug_dump_recv_state: Receive State[0]: src ID 0, addr 00:00:00:00:00:00 (public), adv_sid 0, broadcast_id 1193046, pa_sync_state 0, encrypt state 0, num_subgroups 0
HL [00:00:05.379,852] <dbg> bt_bap_scan_delegator: write_control_point: Data
                                                   04 00 30 31 32 33 34 35  36 37 38 39 41 42 43 44 |..012345 6789ABCD
                                                   45 46                                            |EF               
HL [00:00:05.379,852] <dbg> bt_bap_scan_delegator: write_control_point: Assistant setting broadcast code
HL [00:00:05.379,913] <dbg> bt_bap_scan_delegator: scan_delegator_broadcast_code: Index 0: broadcast code added: 30313233343536373839414243444546
HL [00:00:05.379,913] <inf> bis_headset: BASS recv 0 ID 0x123456 broadcast code
HL [00:00:05.380,187] <dbg> bt_bap_scan_delegator: write_control_point: Data
                                                   03 00 01 ff ff 00                                |......           
HL [00:00:05.380,218] <dbg> bt_bap_scan_delegator: write_control_point: Assistant modifying source
HL [00:00:05.380,249] <dbg> bt_bap_scan_delegator: scan_delegator_mod_src: src_id 0
HL [00:00:05.380,249] <inf> bis_headset: BASS recv 0 ID 0x123456 pa sync
HL [00:00:05.559,936] <dbg> bt_bap_scan_delegator: recv_state_cfg_changed: value 0x0000
HL [00:00:05.560,119] <dbg> bt_bap_scan_delegator: scan_delegator_disconnected: Instance 0 with addr 14:D4:F9:A0:45:33 (public) disconnected
HL [00:00:05.560,241] <inf> bis_headset: Disconnected: 14:D4:F9:A0:45:33 (public) (reason 0x16)
HL [00:00:05.560,882] <inf> bis_headset: Scanning successfully started
```

Client:
```
GW [00:00:15.312,072] <inf> cis_gateway: Device found
GW [00:00:15.414,154] <inf> cis_gateway: Connected: E9:95:8A:50:69:39 (random)
GW [00:00:15.920,379] <inf> ble: Passkey accepted for E9:95:8A:50:69:39 (random): 942432
GW [00:00:16.460,449] <inf> ble: Pairing completed: E9:95:8A:50:69:39 (random), bonded: 1
GW [00:00:16.461,395] <inf> cis_gateway: Updated to 2M PHY
GW [00:00:18.798,614] <inf> cis_gateway: SIRKs discovered: 1
GW [00:00:18.978,637] <dbg> bt_bap_broadcast_assistant: service_discover_func: [ATTRIBUTE] handle 0x000E
GW [00:00:19.158,691] <dbg> bt_bap_broadcast_assistant: char_discover_func: [ATTRIBUTE] handle 0x000F
GW [00:00:19.158,721] <dbg> bt_bap_broadcast_assistant: char_discover_func: Control Point
GW [00:00:19.158,721] <dbg> bt_bap_broadcast_assistant: char_discover_func: [ATTRIBUTE] handle 0x0011
GW [00:00:19.158,752] <dbg> bt_bap_broadcast_assistant: char_discover_func: Receive State 1
GW [00:00:19.518,676] <dbg> bt_bap_broadcast_assistant: char_discover_func: Found 1 BASS receive states
GW [00:00:19.878,692] <inf> cis_gateway: BASS recv discovered: 1
GW [00:00:20.058,746] <dbg> bt_bap_broadcast_assistant: notify_handler: Receive state notification:
                                                        00 00 00 00 00 00 00 00  00 56 34 12 00 00 00    |........ .V4.... 
GW [00:00:20.058,776] <inf> cis_gateway: BASS recv state: src_id 0 sid 0 state 0/0
GW [00:00:20.328,857] <inf> cis_gateway: Disconnected: E9:95:8A:50:69:39 (random) (reason 0x13)
GW [00:00:20.329,589] <inf> cis_gateway: Scanning successfully started
GW [00:00:38.557,678] <inf> cis_gateway: Device found
GW [00:00:38.667,907] <inf> cis_gateway: Connected: E5:CB:14:1C:FD:2A (random)
GW [00:00:39.141,235] <inf> ble: Passkey accepted for E5:CB:14:1C:FD:2A (random): 717060
GW [00:00:39.681,488] <inf> ble: Pairing completed: E5:CB:14:1C:FD:2A (random), bonded: 1
GW [00:00:39.682,495] <inf> cis_gateway: Updated to 2M PHY
GW [00:00:42.019,500] <inf> cis_gateway: SIRKs discovered: 1
GW [00:00:42.199,523] <dbg> bt_bap_broadcast_assistant: service_discover_func: [ATTRIBUTE] handle 0x000E
GW [00:00:42.379,577] <dbg> bt_bap_broadcast_assistant: char_discover_func: [ATTRIBUTE] handle 0x000F
GW [00:00:42.379,577] <dbg> bt_bap_broadcast_assistant: char_discover_func: Control Point
GW [00:00:42.379,608] <dbg> bt_bap_broadcast_assistant: char_discover_func: [ATTRIBUTE] handle 0x0011
GW [00:00:42.379,608] <dbg> bt_bap_broadcast_assistant: char_discover_func: Receive State 1
GW [00:00:42.739,562] <dbg> bt_bap_broadcast_assistant: char_discover_func: Found 1 BASS receive states
GW [00:00:43.099,548] <inf> cis_gateway: BASS recv discovered: 1
GW [00:00:43.279,632] <dbg> bt_bap_broadcast_assistant: notify_handler: Receive state notification:
                                                        00 00 00 00 00 00 00 00  00 56 34 12 00 00 00    |........ .V4.... 
GW [00:00:43.279,632] <inf> cis_gateway: BASS recv state: src_id 0 sid 0 state 0/0
GW [00:00:43.549,743] <inf> cis_gateway: Disconnected: E5:CB:14:1C:FD:2A (random) (reason 0x13)
GW [00:00:43.550,537] <inf> cis_gateway: Scanning successfully started
```

**Environment (please complete the following information):**
 - OS: MacOS build, Nordic nRF5340 targets
 - Toolchain: Zephyr SDK
 - Commit SHA: 28a3fca

Chris-StJohn-Bose, 2023-08-31:
Here's the patch I'm successfully using, but as I said in the description this simply disables the (most of) the callbacks so we need to decide if those callbacks should be removed.
```
diff --git a/subsys/bluetooth/audio/bap_broadcast_assistant.c b/subsys/bluetooth/audio/bap_broadcast_assistant.c
index 7e586d2b8e..3949025b8d 100644
--- a/subsys/bluetooth/audio/bap_broadcast_assistant.c
+++ b/subsys/bluetooth/audio/bap_broadcast_assistant.c
@@ -500,6 +500,11 @@ static int bt_bap_broadcast_assistant_common_cp(struct bt_conn *conn,
                return -EINVAL;
        }

+#if 1
+       err = bt_gatt_write_without_response(conn,
+                                            broadcast_assistant.cp_handle,
+                                            buf->data, buf->len, false);
+#else
        broadcast_assistant.write_params.offset = 0;
        broadcast_assistant.write_params.data = buf->data;
        broadcast_assistant.write_params.length = buf->len;
@@ -510,7 +515,7 @@ static int bt_bap_broadcast_assistant_common_cp(struct bt_conn *conn,
        if (err == 0) {
                broadcast_assistant.busy = true;
        }
-
+#endif
        return err;
 }
```

Thalley, 2023-08-31:
> Here's the patch I'm successfully using, but as I said in the description this simply disables (most of) the callbacks so we need to decide if those callbacks should be removed.
> 
> ```
> diff --git a/subsys/bluetooth/audio/bap_broadcast_assistant.c b/subsys/bluetooth/audio/bap_broadcast_assistant.c
> index 7e586d2b8e..3949025b8d 100644
> --- a/subsys/bluetooth/audio/bap_broadcast_assistant.c
> +++ b/subsys/bluetooth/audio/bap_broadcast_assistant.c
> @@ -500,6 +500,11 @@ static int bt_bap_broadcast_assistant_common_cp(struct bt_conn *conn,
>                 return -EINVAL;
>         }
> 
> +#if 1
> +       err = bt_gatt_write_without_response(conn,
> +                                            broadcast_assistant.cp_handle,
> +                                            buf->data, buf->len, false);
> +#else
>         broadcast_assistant.write_params.offset = 0;
>         broadcast_assistant.write_params.data = buf->data;
>         broadcast_assistant.write_params.length = buf->len;
> @@ -510,7 +515,7 @@ static int bt_bap_broadcast_assistant_common_cp(struct bt_conn *conn,
>         if (err == 0) {
>                 broadcast_assistant.busy = true;
>         }
> -
> +#endif
>         return err;
>  }
> ```

@kruithofa is looking into this issue. The fix may be as simple as moving the `busy = false` to before the callbacks, but since the recv_state callback is called both during discovery and when we receive a notification, it may not be that simple.

kruithofa, 2023-09-26:
@Chris-StJohn-Bose 

Although it is possible to use a ‘Write No Response’ this has 2 drawbacks.
1.	There is no notification in case the write fails
2.	‘Write No Response’ can not be used to write data larger than the MTU size
The relevant part of the spec is in BT Core spec 5.4, vol 3, part G, 4.9.4
 
We looked into moving the busy=false statement, but that didn’t give the desired result.
A better solution is to wait for the write response before doing the ‘Set Broadcast Code’ instead of doing this in the callback. 
 
I suggest to close the issue as a ‘will not do’

Chris-StJohn-Bose, 2023-09-29:
@kruithofa 
> 1. There is no notification in case the write fails

Write failure is a pretty unexpected event really and suggests an implementation or connection problem rather than a state problem? One could equally argue that Notification is the wrong choice.

I'm thinking about similarities with ASCS, for instance.

> 2. ‘Write No Response’ can not be used to write data larger than the MTU size
>    The relevant part of the spec is in BT Core spec 5.4, vol 3, part G, 4.9.4

Fair point. I think BAP requires an MTU of at least 64 octets presumably for the same reason.

> We looked into moving the busy=false statement, but that didn’t give the desired result. A better solution is to wait for the write response before doing the ‘Set Broadcast Code’ instead of doing this in the callback.

I don't think it's reasonable to expect the application to sort out these things. One approach might be to queue callbacks whilst busy==true which would probably be the least surprising behaviour for the application implementer. We still have the problem of potentially two threads working the API at the same time (the UI thread and presumably BT_RX) but hey that's callbacks for you!

Thalley, 2023-09-29:
> Write failure is a pretty unexpected event really and suggests an implementation or connection problem rather than a state problem? One could equally argue that Notification is the wrong choice.

Unless the server rejects your request :) 

> I'm thinking about similarities with ASCS, for instance.

In ASCS all writes are "responded" to by notifications. BASS does not have feature. BASS even defines it own ATT error messages, so arguably it has been designed with write (with response) in mind. 

> Fair point. I think BAP requires an MTU of at least 64 octets presumably for the same reason.

Indeed, but it doesn't take much metadata to get past the 64 octets. 

> I don't think it's reasonable to expect the application to sort out these things. One approach might be to queue callbacks whilst busy==true which would probably be the least surprising behaviour for the application implementer. 

The problem is that in order to support long writes, we'll need to store a buffer of up to 512 octets _per_ pending write request. The challenge is essentially a memory issue. 

In order to get past those memory issue, we'd have to rework the API so that the write buffer is stored in the application, which would move the issue from the stack, to the application (which arguably is better handled for it).

Chris-StJohn-Bose, 2023-10-02:
Ok @Thalley and @kruithofa I guess closing this ticket is probably the right way to go unless a common solution emerges that could be integrated into the API.

I'll probably rework my Broadcast Assistant application to queue recv() callbacks whilst busy==true as the data for these callbacks will be relatively short (they're derived from a notification so can't be that much data in there!)

So we're going with WILL NOT DO -- feel free to re-open if appropriate :-)

Thanks for both of your time looking at this.

Thalley, 2023-10-02:
> (they're derived from a notification so can't be that much data in there!)

The notifications won't even be > GATT MTU, but they may be truncated, in which case you'd need to re-read the receive state. That is a feature still missing from the Broadcast Assistant implementation (will probably use the same implementation of supporting this as we have in the BAP Unicast Client).

closed 2023-10-02 by Thalley