zephyrproject-rtos/zephyr #61717: Occasional TLS CI failures suggest race

d3zd3z, 2023-08-21:
We are seeing occasional failures in various tests that perform networking over TLS. I will update with specific instances once I see them, but most of the time the tests are rerun.

This suggests that there is some kind of race in the TLS code. Discussing with the Mbed TLS team, we may have made an assumption about thread safety of Mbed TLS that isn't true. Apparently, there are no guarantees of thread safety. Recent changes to mbed TLS might have made it a bit more sensitive to these issues, which might by why there is an increase in the occurrence of these errors. A fix might have to re-evaluate how the TLS stack is used from within the Zephyr networking layer to avoid calling the stack from multiple threads.

fabiobaltieri, 2023-08-22:
@d3zd3z setting as low initially, feel free to change it.

d3zd3z, 2023-08-30:
Adding @rlubos and @jukkar for visibility. I'll add another comment with specifics of how I can recreate this.

d3zd3z, 2023-08-30:
I am able to recreate this with:
```bash
west build -p always -b mps2_an521_ns tests/net/socket/tls/
west build -t run
```
Usually one of the tests will deadlock.

I also see the failures on qemu_x86_64.

rlubos, 2023-08-31:
As for `qemu_x86_64` - from @jukkar logs, the long execution time of `test_v4_dtls_sendmsg` suggested that DTLS handshake timed out, I haven't investigated too deep, but increasing mbed TLS heap (`CONFIG_MBEDTLS_HEAP_SIZE=32000`) made the test suite pass.

As for `mps2_an521_ns` - I wonder if we're not hitting here the same issue as with `mps2_an385`, which has been proven to have an unstable system timer (https://github.com/zephyrproject-rtos/zephyr/issues/48608). At least I was able to reproduce the failure of `tests/kernel/timer/timer_monotonic` on `mps2_an521_ns`, by following the steps described in that issue.

carlescufi, 2023-10-04:
Duplicate of #48608

closed 2023-10-04 by carlescufi