zephyrproject-rtos/zephyr #61720: https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/net/cloud/tagoio_http_post

SinghKarminder, 2023-08-22:
Dear Developers

I want to send an image file to TagoIO Server by modifying the repository made by you (Zephyr/samples/net/cloud/tagoio_http_post).

The following function **"tagoio_http_push"** can only send the text data to tagoio server.

**Function: **tagoio_http_push****
**int tagoio_http_push(struct tagoio_context *ctx, http_response_cb_t resp_cb)
{
	struct http_request req;
	int ret;

	memset(&req, 0, sizeof(req));
	long int res = collect_data();
	const char *image_path = data.txt;

	req.method = HTTP_POST;
	req.host = TAGOIO_SERVER;
	req.port = HTTP_PORT;
	req.url = TAGOIO_URL;
	req.header_fields = tagoio_http_headers;
	req.protocol = "HTTP/1.1";
	req.response = resp_cb;
	req.payload		= data.txt;	
	req.payload_len		= res;
	req.recv_buf		= ctx->resp;
	req.recv_buf_len	= sizeof(ctx->resp);

	ret = http_client_req(ctx->sock, &req,
			      CONFIG_TAGOIO_HTTP_CONN_TIMEOUT * MSEC_PER_SEC,
			      ctx);

	close(ctx->sock);
	ctx->sock = -1;

	return ret;
}**

The function for ctx-> payload is as follows:
**Function:
static int collect_data(void)
{
	#define lower 20000
	#define upper 100000
	#define base  1000.00f

	float temp;

	/* Generate a temperature between 20 and 100 celsius degree */
	temp = ((sys_rand32_get() % (upper - lower + 1)) + lower);
	temp /= base;

	(void)snprintf(ctx.payload, sizeof(ctx.payload),
		       "{\"variable\": \"temperature\","
		       "\"unit\": \"c\",\"value\": %f}",
		       (double)temp);

	/* LOG doesn't print float #18351 */
	LOG_INF("Temp: %f", (float) temp);

	return 0;
}**

But I want to send the image file to TagoIo Server.
Can you please suggest the required changes that can I made to send the image file to TagoIo Server.
Is there any separate function is required to push the image.
Thanks in advance.

nandojve, 2023-08-22:
Hi @SinghKarminder ,

You can find all information about how to use device send data at Tago.IO documentation. The explanation is there about the API and their endpoints. I recommend you look for Arduino samples which are quite simple and easy to follow. You can find more info about HTTP [here](https://developer.mozilla.org/en-US/docs/Web/HTTP/Overview), for instance. The current application sends simple raw data to some endpoint and I don't want to change that. You are free to submit another PR based the current implementation which shows how to submit an image. I think it is better not mix topics.

If you dedicate your effort to develop using a PC to cloud the process it will be easy to figure out all needs. Once you understood all details and how your application should like it won't be difficult to replicate on Zephyr.

SinghKarminder, 2023-08-23:
Hi @nandojve,

Thanks for your response. I will try to follow your instructions.
Thanks once again for giving time to my problem.