zephyrproject-rtos/zephyr #61721: Contributor Nomination benediktibk

benediktibk, 2023-08-22:
I'll nominate hereby myself for the triage permissions, as @MaureenHelm pointed me towards this direction. The main reason to do so is because sometimes I might want to add the dev-review label to a PR so that it can be discussed in the dev review meeting.

# Nomination

## GitHub User

Provide the following information about the GitHub user:

1. Full Name: Benedikt Schmidt
1. GitHub username: benediktibk
1. Organization (optional): SILA Embedded Solutions GmbH

## Supporting Documents

https://github.com/zephyrproject-rtos/zephyr/pull/59999
https://github.com/zephyrproject-rtos/zephyr/pull/60329
https://github.com/zephyrproject-rtos/zephyr/pull/59763
https://github.com/zephyrproject-rtos/zephyr/pull/57730
https://github.com/zephyrproject-rtos/zephyr/pull/56234
https://github.com/zephyrproject-rtos/zephyr/pull/54911
https://github.com/zephyrproject-rtos/zephyr/pull/54565

[TSC Project Roles]: <https://docs.zephyrproject.org/latest/project/project_roles.html>
[GitHub Permission Level]: <https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-roles-for-an-organization>

nashif, 2023-08-22:
https://github.com/zephyrproject-rtos/zephyr/invitations

nashif, 2023-08-22:
Thank you for your contributions.

closed 2023-08-22 by nashif