zephyrproject-rtos/zephyr #61725: Incremental twister builds (--no-clean option) result in build error

martinjaeger, 2023-08-22:
**Describe the bug**

Subsequent twister runs with the below option fail if the `twister-out` dir is not removed.

```
  -n, --no-clean        Re-use the outdir before building. Will result in
                        faster compilation since builds will be incremental.
```

Not sure if it's a general twister issue or if it just happens for the ISO-TP tests.

**To Reproduce**

Steps to reproduce the behavior:

1. `rm -rf twister-out*`
2. Run `scripts/twister -T tests/subsys/canbus/isotp -n` for the first time. The tests will succeed.
3. Run `scripts/twister -T tests/subsys/canbus/isotp -n` a second time. The tests fail.

**Expected behavior**

Incremental builds should work reliably.

**Impact**

Subsequent builds during development take longer.

**Logs and console output**

Successful first build:
```
ZEPHYR_BASE unset, using "xxx/zephyr"
Install the anytree module to use the --test-tree option
Install tabulate python module with pip to use --device-testing option.
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.4.0-2584-gd4e33c570835
INFO    - Using 'zephyr' toolchain.
INFO    - Selecting default platforms per test case
INFO    - Building initial testsuite list...
INFO    - Writing JSON report xxx/zephyr/twister-out/testplan.json
INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:   84/  84  100%  skipped:   83, failed:    0, error:    0
INFO    - 2 test scenarios (84 test instances) selected, 83 configurations skipped (79 by static filter, 4 at runtime).
INFO    - 1 of 84 test configurations passed (100.00%), 0 failed, 0 errored, 83 skipped with 0 warnings in 10.86 seconds
INFO    - In total 14 test cases were executed, 994 skipped on 42 out of total 602 platforms (6.98%)
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report xxx/zephyr/twister-out/twister.json
INFO    - Writing xunit report xxx/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report xxx/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```

Failing second build:
```
ZEPHYR_BASE unset, using "xxx/zephyr"
Install the anytree module to use the --test-tree option
Install tabulate python module with pip to use --device-testing option.
Keeping artifacts untouched
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.4.0-2584-gd4e33c570835
INFO    - Using 'zephyr' toolchain.
INFO    - Selecting default platforms per test case
INFO    - Building initial testsuite list...
INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
ERROR   - native_posix              tests/subsys/canbus/isotp/conformance/canbus.isotp.conformance  ERROR : Build failure
ERROR   - see: xxx/zephyr/twister-out/native_posix/tests/subsys/canbus/isotp/conformance/canbus.isotp.conformance/handler.log
INFO    - Total complete:   84/  84  100%  skipped:   83, failed:    0, error:    1
INFO    - 2 test scenarios (84 test instances) selected, 83 configurations skipped (79 by static filter, 4 at runtime).
INFO    - 0 of 84 test configurations passed (0.00%), 0 failed, 1 errored, 83 skipped with 0 warnings in 7.59 seconds
INFO    - In total 15 test cases were executed, 993 skipped on 42 out of total 602 platforms (6.98%)
INFO    - 0 test configurations executed on platforms, 1 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report xxx/zephyr/twister-out/twister.json
INFO    - Writing xunit report xxx/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report xxx/zephyr/twister-out/twister_report.xml...
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - The following issues were found (showing the top 10 items):
INFO    - 1) tests/subsys/canbus/isotp/conformance/canbus.isotp.conformance on native_posix error (Build failure)
INFO    - 
INFO    - To rerun the tests, call twister using the following commandline:
INFO    - west twister -p <PLATFORM> -s <TEST ID>, for example:
INFO    - 
INFO    - west twister -p native_posix -s tests/subsys/canbus/isotp/conformance/canbus.isotp.conformance
INFO    - or with west:
INFO    - west build -p -b native_posix -T tests/subsys/canbus/isotp/conformance/canbus.isotp.conformance
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - Run completed
```

**Environment (please complete the following information):**

 - OS: Manjaro Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: d4e33c570835d12798c17723ebdb371bab9dfe82

nashif, 2023-08-25:
This seems to be the case for `native_posix` only. not sure is the issue yet, it fails when linking... 

@aescolar  any idea?

aescolar, 2023-08-25:
> This seems to be the case for native_posix only. 

I have seen this happening for a few months, and I have never digged into why it happens. It has nothing to do with twister, but cmake. (I don't think it happened ~1 year ago)
You can see happen by calling cmake again.
For ex:
```
build$ cmake ../samples/hello_world/ -DBOARD=native_posix -GNinja 
[...]
build$ ninja
[1/88] Generating include/generated/version.h
-- Zephyr version: 3.4.99 (/scratch/zephyrproject/zephyr), build: zephyr-v3.4.0-2767-g3a4bebacb143
[88/88] Linking C executable zephyr/zephyr.elf
build$ cmake ../samples/hello_world/ -DBOARD=native_posix -GNinja 
[...]
$ ninja
[3/3] Linking C executable zephyr/zephyr.elf
/usr/bin/ld.bfd: app/libapp.a(main.c.obj): warning: relocation in read-only section `.text._posix_zephyr_main'
/usr/bin/ld.bfd: warning: creating DT_TEXTREL in a PIE
```

aescolar, 2023-08-25:
@tejlmand ^^

nashif, 2023-08-25:
did some bisecting, hit this 2a76637963d510583365f45ab7a7ef9b297ecab9 @dcpleung  can you please look into this?

dcpleung, 2023-08-25:
Hm... it's weird. After the second cmake run, `-Wl,-no-pie` is no longer being passed during linking. This happens on Ubuntu 22.04 LTS, but not on Gentoo (GCC 11 vs 12, and cmake 3.27.4 vs 3.27.3).

PR #61912 (2023-09-01): cmake: modules/FindGnuLd: cache variable GNULD_VERSION_STRING
This changes to cache variable GNULD_VERSION_STRING across cmake runs. This variable is used to determine whether -Wl,-no-pie is being passed to linker. However, if cmake is run multiple times without clearing the build directory, GNULD_VERSION_STRING was lost and the script falsely assumed the linker could not take this argument, and thus omitting it during linking. Depending on the host, it would warn on something like this:

  /usr/bin/ld.bfd: app/libapp.a(main.c.obj): warning: relocation in read-only section `.text._posix_zephyr [... cut]
commit 2d78cb74 (2023-09-01): cmake: modules/FindGnuLd: cache variable GNULD_VERSION_STRING

This changes to cache variable GNULD_VERSION_STRING across cmake
runs. This variable is used to determine whether -Wl,-no-pie is
being passed to linker. However, if cmake is run multiple times
without clearing the build directory, GNULD_VERSION_STRING was
lost and the script falsely assumed the linker could not take
this argument, and thus omitting it during linking. Depending on
the host, it would warn on something like this:

  /usr/bin/ld.bfd: app/libapp.a(main.c.obj):
      warning: relocation in read-only section
      `.text. [... cut]
closed 2023-09-01 by nashif