zephyrproject-rtos/zephyr #61741: EATT does not play well with MTU values

daee-demant, 2023-08-22:
** Brief **
The bt_att_get_mtu api uses the maximum mtu when given multiple channels. This is not compatible with having credit based and non-credit based channels.

** Elaborate extensively: **
We have a scenario where we start up by forming an ATT and then proceed to form an EATT.

The net result of this procedure is that we get an extra NULL frame whenever we get a callback from gatt_read_rsp.

We do the following:
1) We establish a Maximum Payload Size (MPS) of 247. 
2) We negotiate an MTU for our ATT with a value of 247.
3) We establish an EATT with the same MPS. This gives an MTU of 245 to leave space for the SDU size.
4) On the EATT we attempt to read a characteristic of size 512.

* What should have happened: *
In the function gatt_read_rsp, our callback function is called by the call

if (params->func(conn, 0, params, pdu, length) == BT_GATT_ITER_STOP)

Our callback determines that the packet has not ended and transmits a BT_GATT_ITER_CONTINUE

We have a condition in the next line to determine if we have ended.

if (length < bt_att_get_mtu(conn) - 1)) {

This decides by not triggering that we have used a full MTU length and therefore starts a blob read request for the remainder of the packet.

We read a second chunk with the same path.

And finally we read 24 bytes, which is less than 244, wherefore the end condition is triggered, a NULL packet is sent to the callback, and we can end the reading.

* What actually happens: *
The bt_att_get_mtu now iterates through the available channels and picks the maximum MTU, which is a maximum of the 247 of the ATT and the 245 of the EATT. Therefore the comparison now becomes 

if ([244] < [247] - 1)) {

which will always trigger, even if the packet is full, effectively making the code below it dead.

** Suggestion: **
Since the MTU can be different from channel to channel and this is a scenario where we should be using the MTU of a specific channel, it would seem to make sense to have an API equivalent to bt_att_get_mtu, but for channels rather than connections.

carlescufi, 2023-08-24:
Bluetooth WG summary:

- Document properly how to use `bt_gatt_read()` to select between regular and blob/long reads
- Fix the MTU check for EATT in `att_internal.h` (in `att_chan` we will add a `req_mtu`) and then use that in `att.c` in `gatt_read_rsp`

daee-demant, 2023-10-10:
Apologies, I've been on vacation for the last 3 weeks. Yes, I think it looks like a well thought out solution.

closed 2023-10-23 by carlescufi