zephyrproject-rtos/zephyr #61761: Cooperative Threads are being Preempted via context switches in _HandlerModeExit

csheaupdesigns, 2023-08-22:
**Describe the bug**
The function _HandlerModeExit in zephyr/arch/arm/core/aarch32/cortex_m/exc_exit.S appears to ignore cooperative thread priority when deteriming whether or not a context switch is required.  The following excerpt is from the Zephyr Architecture Porting Guide:

_In this case, the context switch must only be invoked when the interrupted thread was preemptible, not when it was a cooperative one, and only when the current interrupt is not nested.

_The kernel also has the concept of “locking the scheduler”. This is a concept similar to locking the interrupts, but lighter-weight since interrupts can still occur. If a thread has locked the scheduler, is it temporarily non-preemptible.

So, the decision logic to invoke the context switch when exiting an interrupt is simple:

If the interrupted thread is not preemptible, do not invoke it.
Else, fetch the cached thread from the ready queue, and:
If the cached thread is not the current thread, invoke the context switch.
Else, do not invoke it.__

However, in _HandlerModeExit, the logic only looks to see if the cached thread is not the current thread and does not check the priority of the current thread to determine if it is cooperative.  This can sometimes result in a context switch from within a cooperative thread when an ISR preempts it.  In my application, it is currently breaking the functionality of the scheduler lock.

Please also mention any information which could help others to understand
the problem you're facing:
 - I am developing on a Nordic nrf5340.
 - I have not yet tried adding logic to include checking of the priority to verify it resolves my issue.
 - ...

**To Reproduce**
Steps to reproduce the behavior:
-Just run code within a cooperative thread with periodic interrupts occurring.  This logic executes on every ISR unless using the direct ISR method within Zephyr.

**Expected behavior**
I expected this function to only perform a context switch if returning to a preemptible thread AND the cached thread is not the current thread.

**Impact**
I am unable to use the scheduler lock functionality, which is important to the application I am working on.

**Environment (please complete the following information):**
 - The bug is present in the latest version of Zephyr (3.4.99)

csheaupdesigns, 2023-08-23:
I should clarify in my steps to reproduce the error that I am running in a loop in one thread that is locking the scheduler, delaying 5 ms, and then unlocking.  During the delay period, I am not doing anything that would deem the thread "unready".  I'm noting this because I could envision a case in which the _kernel_offset_to_ready_q_cache was updated to a thread other than the current thread, then the scheduler was locked, and then the interrupt occurs.  In this case, the scheduler would have been correct to update the _kernel_offset_to_ready_q_cache when it ran the logic.  However, it should still check the priority of the current thread from within the interrupt handler prior to making the context switch.

csheaupdesigns, 2023-10-30:
Please remove the stale label.  I have not seen any work completed to fix the issue and there have been no discussions other than my original report and comments.

csheaupdesigns, 2024-01-02:
Remove stale label

andyross, 2024-01-02:
Saw this got resurrect.  I'm not really the right assignee here.  This is in the arm32 layer, not the kernel.    ARM32 is a pre-switch architecture which doesn't call z_get_next_switch_handle() on its interrupt exit path and is responsible for getting this right on its own.

ithinuel, 2024-01-30:
If my understanding of the issue is correct, the change in 2659343d203b554640e1da623340b8bd812b07d4 should fix it.

This PR converts some of the non-essential assembly code to C to drastically ease the maintenance.

csheaupdesigns, 2024-02-15:
I am unable to test the fix at this time but I can verify that the commit notes and the source code do appear to address the problem.  I will close the case.

closed 2024-02-15 by csheaupdesigns

kietavainen, 2025-01-29:
> Saw this got resurrect. I'm not really the right assignee here. This is in the arm32 layer, not the kernel. ARM32 is a pre-switch architecture which doesn't call z_get_next_switch_handle() on its interrupt exit path and is responsible for getting this right on its own.

@andyross, Can you confirm if `_kernel.ready_q.cache != _kernel.cpus->current` before the context switch is enough to get this right? That's the only condition there used to be before 42036cd. AFAIK, `_kernel.ready_q.cache` is the next thread to run, and if that's not the current thread, a context switch is needed. If the current thread is cooperative, the scheduler would not update the cached thread unless a meta-IRQ should preempt it.

The Architecture Porting Guide hints that more checking would be needed, but is it still up to date? It doesn't mention meta-IRQs.
https://github.com/zephyrproject-rtos/zephyr/blob/c66a0bb2a9269dc483079d5cc24ba77b8c332a9a/doc/hardware/porting/arch.rst?plain=1#L248-L257

I'm wondering if reverting 42036cd is the right thing to do to fix #80574. The check added in it doesn't consider preemptible threads that have locked the scheduler, nor meta-IRQs:
https://github.com/zephyrproject-rtos/zephyr/blob/42036cdbcae270d00b5bd08e805e33bbd3b5787f/arch/arm/core/cortex_m/exc_exit.c#L58-L59

If these checks are needed, maybe something like this should be used instead:
```
--- a/arch/arm/core/cortex_m/exc_exit.c
+++ b/arch/arm/core/cortex_m/exc_exit.c
@@ -55,8 +55,8 @@ FUNC_ALIAS(z_arm_exc_exit, z_arm_int_exit, void);
 Z_GENERIC_SECTION(.text._HandlerModeExit) void z_arm_exc_exit(void)
 {
 #ifdef CONFIG_PREEMPT_ENABLED
-       /* If thread is preemptible */
-       if (_kernel.cpus->current->base.prio >= 0) {
+       /* If thread is preemptible or the cached thread is a meta-IRQ */
+       if (thread_is_preemptible(_kernel.cpus->current) || thread_is_metairq(_kernel.ready_q.cache)) {
                /* and cached thread is not current thread */
                if (_kernel.ready_q.cache != _kernel.cpus->current) {
                        /* trigger a context switch */
```