zephyrproject-rtos/zephyr #61766: GPIO hog using gpio expander PCAL5408A

SGilbert-bp, 2023-08-23:
Using gpio-hog option for several pins with the gpio expander driver for PCAL5408A doesn't appear to set the GPIOs up as expected. 

On further investigation, the hog_init function is called before the gpio expander driver init function is called leading to a device not ready error.

Current work around is to change the priority of the gpio_hogs priority from 41(42?) to something higher than the PCAL54xxA driver (70) 

**Expected behavior**
IO expander to be initialised with pins set to devicetree defined initial configuration.

**Impact**
can't use default priorities for this GPIO expander and default pin configuration specced in devicetree. Minor issue but took a while to track down

danielstuart14, 2023-08-24:
@henrikbrixandersen does it make sense to create a PR to change the default gpio hog init priority (I just checked all the GPIO drivers, the maximum value is 99, so we could set it to an even 100), or do you think it's best to let the user change it manually through kconfig if using an expander? 

My main concern with changing the default priority is the increase in delay for the gpio hog configuration to be applied at boot, mostly when using native gpio drivers.

henrikbrixandersen, 2023-08-24:
> @henrikbrixandersen does it make sense to create a PR to change the default gpio hog init priority (I just checked all the GPIO drivers, the maximum value is 99, so we could set it to an even 100), or do you think it's best to let the user change it manually through kconfig if using an expander?

I think it is best left up to the user to change it.

> My main concern with changing the default priority is the increase in delay for the gpio hog configuration to be applied at boot, mostly when using native gpio drivers.

A valid concern, I'd say.

SGilbert-bp, 2023-08-24:
Looks like I should close this issue but before I do, Is there a mechanism for raising a warning if hogging is enabled but has a higher priority than a gpio expander?

henrikbrixandersen, 2023-08-24:
> Looks like I should close this issue but before I do, Is there a mechanism for raising a warning if hogging is enabled but has a higher priority than a gpio expander?

Not at the moment, I'm afraid, since a GPIO hog currently falls outside of our devicetree dependency check.
The implementation is planned to change, though, so that GPIO hogs will automatically be initialized along with their corresponding GPIO controller, instead of all at once.

SGilbert-bp, 2023-08-24:
Thanks for clarifying

closed 2023-08-24 by SGilbert-bp