zephyrproject-rtos/zephyr #61783: [Backport v3.4-branch] Failed to backport #61398

zephyrbot, 2023-08-23:
This issue was created automatically because the backport of #61398 to `v3.4-branch` failed.

The backport to `v3.4-branch` failed:
```
Auto-merging include/zephyr/posix/posix_types.h
Auto-merging include/zephyr/posix/signal.h
CONFLICT (content): Merge conflict in include/zephyr/posix/signal.h
Auto-merging lib/posix/timer.c
error: could not apply 9f1e4fa86c... posix: signal: type corrections for sigval, sigevent, notify attr
hint: After resolving the conflicts, mark them with
hint: "git add/rm <pathspec>", then run
hint: "git cherry-pick --continue".
hint: You can instead skip this commit with "git cherry-pick --skip".
hint: To abort and get back to the state before "git cherry-pick",
hint: run "git cherry-pick --abort".
```
To backport manually, run these commands in your terminal:
```bash
# Fetch latest updates from GitHub
git fetch
# Create a new working tree
git worktree add .worktrees/backport-v3.4-branch v3.4-branch
# Navigate to the new working tree
cd .worktrees/backport-v3.4-branch
# Create a new branch
git switch --create backport-61398-to-v3.4-branch
# Cherry-pick the merged commits of this pull request and resolve the conflicts
git cherry-pick -x e6c6d82a00922e1b29a72611a802122564c5b9ba~3..e6c6d82a00922e1b29a72611a802122564c5b9ba
# Push it to GitHub
git push --set-upstream origin backport-61398-to-v3.4-branch
# Go back to the original working tree
cd ../..
# Delete the working tree
git worktree remove .worktrees/backport-v3.4-branch
```
Then, create a pull request where the `base` branch is `v3.4-branch` and the `compare`/`head` branch is `backport-61398-to-v3.4-branch`.

henrikbrixandersen, 2023-10-23:
@cfriedt Please follow up on this.

henrikbrixandersen, 2023-10-31:
Closing as https://github.com/zephyrproject-rtos/zephyr/pull/64454 was merged.

PR #64454 (2023-10-27): [v3.4] posix: signal: type corrections for sigval, sigevent, notify attr
Backport of #61398 to `v3.4-branch`

Fixes #61783
Fixes #60957
PR #64273 (2023-10-31): [v3.4] type corrections for sigval sigevent notify
Backport of #61398 to `v3.4-branch`.

Fixes #60957
Fixes #61783
closed 2023-10-31 by henrikbrixandersen