zephyrproject-rtos/zephyr #61818: Fuel Gauge API Improvements

aaronemassey, 2023-08-23:
**Is your enhancement proposal related to a problem? Please describe.**

There's been some time and usage of the fuel gauge API since it's merging. In
that time and during the review of the [Charger Driver API
PR](https://github.com/zephyrproject-rtos/zephyr/pull/56666), some potential
sticking points have been brought up about the fuel gauge API that could use
some attention.

These are documented as the checklist below with a short summary of the
motivation.

- [ ] Multi-property functions use the term "props" instead of "prop"
  - *Why*: This will just be easier to understand as we're not mixing plural
    with singular tense.
- [ ] Iterating over properties during a multi-get/set shall be a separate
      system call defined in the header
	  - [ ] Iterating using a single get/set will be the default with an API
            check for driver improvements.
	  - *Why*: Although enabling a multi-get/set has been requested, it isn't a
        necessarilary common use case and the trivial solution (iterating over
        all the properties) will be common amongst many drivers. We will check
        for a driver specific function in case it has made optimizations.
- [ ] Extract SBS Specific Properties as a separate public header
  - *Why* This allows the use of SBS Fuel gauge properties without cluttering up
    the main API properties
- [ ] Separate exit code out from the input struct
- [ ] Separate value out from the input struct as an output union
  - [ ] At this point, we can replace the input property struct with just the union
  
The assignee of this issue (@aaronemassey) will also be responsible for
migrating drivers as needed.
		
**Additional context**

Part of https://github.com/zephyrproject-rtos/zephyr/issues/44847 work.

**NOTE**

Several of these are API breaking changes. This is acceptable as long as Zephyr
CI passes because this API is currently marked as "experimental".

devbhavej, 2026-01-22:
Cool, thanks for the update. I'll try to create a PR this week.

DBS06, 2026-01-22:
@devbhavej  @aaronemassey As far as I understood this would mean renaming the property enum fields in the [fuel_gauge.h](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/drivers/fuel_gauge.h) similar as in  [charger-h](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/drivers/charger.h), e.g.:

`CHARGER_PROP_CONSTANT_CHARGE_CURRENT_UA`
which means for the fuel_gauge.h e.g.:
`FUEL_GAUGE_VOLTAGE` -> `FUEL_GAUGE_VOLTAGE_UV`

Same goes for the `union fuel_gauge_prop_val`.

Imho this would clarify which unit the property has, which would be for sure an improvement, but the downside is, that this would be of course a breaking change which needs to be handled carefully. The fuel_gauge driver already left the experimental state into the unstable state (https://docs.zephyrproject.org/latest/develop/api/overview.html#api-overview), therefore, afaik breaking changes are possible.

The question would be how we are going to handle this, I think there are two possibilities:

1. Just change name and make a hard cut which results in a breaking change
2. Leave `FUEL_GAUGE_VOLTAGE` in and mark it as deprecated (like [here](https://github.com/zephyrproject-rtos/zephyr/blob/cbc881cc51f10686e04cc14b1ce4fcb05aaf3b15/include/zephyr/bluetooth/conn.h#L1064) ) and additionally add `FUEL_GAUGE_VOLTAGE_UV` and remove the "old" properties later.

@kartben What is your opinion about it?

devbhavej, 2026-01-22:
Thanks @DBS06. I liked the 2nd option better. This will give developers some time to start using the new enums/struct fields. How about the struct fields? Do we just add and use a new field with `*_ua` and then mark the original ones as deprecate? Since it is union, they will get the same value so functionally it should not break anything. 

I agree, I will wait for clarifying details before I start working on this change.

devbhavej, 2026-01-28:
> [@devbhavej](https://github.com/devbhavej) [@aaronemassey](https://github.com/aaronemassey) As far as I understood this would mean renaming the property enum fields in the [...]

Hey @kartben @aaronemassey, any thoughts/opinions on how to approach this?  Thanks!