zephyrproject-rtos/zephyr #61822: RFC: add heatshrink compression library support

RomainPelletant, 2023-08-24:
## Origin

Name of project hosting the original open source code : [Heatshrink](https://github.com/atomicobject/heatshrink)

## Purpose

Compression/Decompression library with low memory footprint.
Decompression takes 800 bytes flash usage and 512 bytes to 2kB of RAM depending on compression level.

[image]
RAM report

[image]
ROM report

## Mode of integration

Offer a low memory footprint compression library to enable OTA or Delta OTA on most constrained platforms.
Heatshrink is used in several Delta processing tentatives for MCU.
Integration as module, named heatshrink.

## Maintainership

List the person(s) that will be maintaining the integration of this external code
for the foreseeable future. Please use GitHub IDs to identify them. You can
choose to identify a single maintainer only or add collaborators as well
- RomainPelletant : [ID](https://github.com/RomainPelletant)
- ClovisCorde : [ID](https://github.com/ClovisCorde)

## Pull Request

[PR](https://github.com/zephyrproject-rtos/zephyr/pull/61821)

## Description

Heatshrink is a low memory footprint compression/decompression library.
It enables on-the-fly decompression, useful to directly save on flash decompression output without the need of consequent buffer length.
That library is used in several Delta update tentatives/thesis and it could be a first step to implement a Delta API.

## Dependencies

Direct calls to the library are preferred.

## Revision

SHA : 7d419e1fa4830d0b919b9b6a91fe2fb786cf3280

## License

``ISC``

nordicjm, 2023-08-24:
Have previously used this library in MCU-based applications and was seeing ~27% average reduction in bin filesize (which was great), the library itself has a small overhead.

nashif, 2023-08-30:
TSC: under review, we are considering to introduce an area for integration optional modules which will allow listing this as something a user can integrate in their application without making it a module maintained in the default manifest.

nashif, 2023-10-19:
this should be done as an external module, i.e. by providing a reference and enabled by those who need it in their manifest, see https://docs.zephyrproject.org/latest/develop/manifest/index.html#external-projects-modules

closed 2023-10-19 by nashif

RomainPelletant, 2023-10-19:
@nashif thanks for your feedback

RomainPelletant, 2023-10-19:
@nashif if I well understood, we shall only update our repository to be included as external module and we shall update documentation to add our external module to replace "TBD".
[image]
Is that array, the expected columns could be: "project", "revision", "url", and maybe a description?