zephyrproject-rtos/zephyr #61823: tests-ci : kernel: scheduler: slice_perthread test Failed

hakehuang, 2023-08-24:
**Describe the bug**
    slice_perthread test is Failed on zephyr-v3.4.0-2584-gd4e33c570835 on mimxrt1170_evk_cm7
    testcase path is kernel/scheduler/slice_perthread/slice_perthread

    see logs for details

    **To Reproduce**
    1. 
    ```
    scripts/twister --device-testing --device-serial /dev/ttyACM0 -p mimxrt1170_evk_cm7      --sub-test kernel.scheduler
    ```
    or
    ```
    # cd tests/kernel/scheduler/slice_perthread/slice_perthread
    # west build -b mimxrt1170_evk_cm7
    # west flash
    ```
    2. See error

    **Expected behavior**
    test pass

    **Impact**

    **Logs and console output**
    
    ```
    START - test_slice_perthread

Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:159: slice_expired: ((dt - PERTHREAD_SLICE_TICKS) <= TICK_SLOP is false)
slice expired >4 ticks late (dt=63)
FAIL - test_slice_perthread in 0.026 seconds

        
    ```

    **Environment (please complete the following information):**
     - OS: (e.g. Linux )
     - Toolchain (e.g Zephyr SDK)
     - Commit SHA or Version used: zephyr-v3.4.0-2584-gd4e33c570835

hakehuang, 2023-08-24:
last weekly zephyr-v3.4.0-2252-gd11c0a1664f6  it is OK, looks like it is a possiblilty issue.

henrikbrixandersen, 2023-08-24:
@hakehuang The way you're submitting these issues somehow messes up the Markdown formatting.

hakehuang, 2023-08-25:
> @hakehuang The way you're submitting these issues somehow messes up the Markdown formatting.

sorry, my templet issue, let me corect them

closed 2023-11-12 by github-actions[bot]