zephyrproject-rtos/zephyr #61824: tests-ci : net: lwm2m: lwm2m_rd_client_dtls test Failed

hakehuang, 2023-08-24:
**Describe the bug**
    lwm2m_rd_client_dtls test is Failed on zephyr-v3.4.0-2584-gd4e33c570835 on lpcxpresso54114_m4
    testcase path is net/lwm2m/lwm2m_rd_client_dtls/suspend_resume_registration

    see logs for details

    **To Reproduce**
    1. 
    ```
    scripts/twister --device-testing --device-serial /dev/ttyACM0 -p lpcxpresso54114_m4      --sub-test net.lwm2m
    ```
    or
    ```
    # cd tests/net/lwm2m/lwm2m_rd_client_dtls/suspend_resume_registration
    # west build -b lpcxpresso54114_m4
    # west flash
    ```
    2. See error

    **Expected behavior**
    test pass

    **Impact**

    **Logs and console output**
    
    ```
    START - test_suspend_resume_registration
Event!
I: Start LWM2M Client: Test
Event!
D: State: 1
Event!
D: State: 2
I: Overwrite a server lifetime with default
I: RD Client started with endpoint 'Test' with client lifetime 60
I: Overwrite a server lifetime with default
D: registration sent [192.168.1.1:4444]
D: Registration callback (code:2.1)
I: **** LWM2M_RD_CLIENT_EVENT_REGISTRATION_COMPLETE
I: Registration Done (EP='jATO2yn9u7')
Event!
D: State: 5
I: **** LWM2M_RD_CLIENT_EVENT_QUEUE_MODE_RX_OFF
D: lwm2m_rd_client_pause()
I: Suspend client
I: **** LWM2M_RD_CLIENT_EVENT_ENGINE_SUSPENDED
I: Expecting 11 at 1, events:
I: [0] = 5
I: [1] = 10
I: [2] = 11

Assertion failed at WEST_TOPDIR/zephyr/tests/net/lib/lwm2m/lwm2m_rd_client/src/main.c:492: lwm2m_rd_client_test_suspend_resume_registration: check_lwm2m_rd_client_event(LWM2M_RD_CLIENT_EVENT_ENGINE_SUSPENDED, 1) is false

Event!
D: State: 9
FAIL - test_suspend_resume_registration in 0.119 seconds

        
    ```

    **Environment (please complete the following information):**
     - OS: (e.g. Linux )
     - Toolchain (e.g Zephyr SDK)
     - Commit SHA or Version used: zephyr-v3.4.0-2584-gd4e33c570835

hakehuang, 2023-08-24:
@dleach02  this case is newly added for lwmem changes

PerMac, 2023-08-25:
We also see this on our boards. I was told by Seppo that the test should only run on simulators, but the current yaml doesn't reflect this. AFAIK we don't have an entry in yaml to allow only on sims.

SeppoTakalo, 2023-08-25:
I don't have this particular board, and honestly, I never ran these tests on HW as they are very sensitive to timing.

Can you re-run now that https://github.com/zephyrproject-rtos/zephyr/pull/59823  is merged?
I changed the RD client tests a bit, so that is should not be that sensitive.

The problem was that how RD client tests have been written, they expect certain event to come at certain timestamp. But if RX_OFF comes first, it might ruin the order. It is not really a bug in a RD client, but just how the event-loop is stubbed in test case.

hakehuang, 2023-08-30:
@SeppoTakalo , this case still failed

```
*** Booting Zephyr OS build zephyr-v3.4.0-2851-g34fddd8b4c09 ***
Running TESTSUITE lwm2m_rd_client
===================================================================
START - test_bootstrap_fail
I: Start LWM2M Client: Test
D: State: 1
D: State: 2
I: Bootstrap started with endpoint 'Test' with client lifetime 0
D: Register ID with bootstrap server as 'ep=Test'
D: Bootstrap callback (code:4.0)
E: Failed with code 4.0 (Bad Request). Not Retrying.
I: *** LWM2M_RD_CLIENT_EVENT_BOOTSTRAP_REG_FAILURE
D: State: 0
 PASS - test_bootstrap_fail in 0.069 seconds
===================================================================
START - test_bootstrap_no_srv_fallback_to_register
I: Start LWM2M Client: Test
D: State: 1
D: State: 2
W: sec_obj_inst: No matching servers found.
W: Bootstrap server not found! Try normal registration.
D: State: 6
I: RD Client started with endpoint 'Test' with client lifetime 60
D: registration sent [192.168.1.1:4444]
D: Registration callback (code:2.1)
I: *** LWM2M_RD_CLIENT_EVENT_REGISTRATION_COMPLETE
I: Registration Done (EP='jATO2yn9u7')
D: State: 9
 PASS - test_bootstrap_no_srv_fallback_to_register in 0.087 seconds
===================================================================
START - test_bootstrap_timeout
I: Start LWM2M Client: Test
D: State: 1
D: State: 2
I: Bootstrap started with endpoint 'Test' with client lifetime 0
D: Register ID with bootstrap server as 'ep=Test'
W: Bootstrap Timeout
I: *** LWM2M_RD_CLIENT_EVENT_BOOTSTRAP_REG_FAILURE
D: State: 1
 PASS - test_bootstrap_timeout in 0.063 seconds
===================================================================
START - test_deregistration_timeout
I: Start LWM2M Client: Test
D: State: 1
D: State: 6
I: RD Client started with endpoint 'Test' with client lifetime 60
D: registration sent [192.168.1.1:4444]
D: Registration callback (code:2.1)
I: *** LWM2M_RD_CLIENT_EVENT_REGISTRATION_COMPLETE
I: Registration Done (EP='jATO2yn9u7')
D: State: 9
I: Stop LWM2M Client: Test
D: State: 14
I: Deregister from 'jATO2yn9u7'
W: De-Registration Timeout
I: *** LWM2M_RD_CLIENT_EVENT_DEREGISTER_FAILURE
D: State: 0
 PASS - test_deregistration_timeout in 0.099 seconds
===================================================================
START - test_engine_trigger_bootstrap
I: Start LWM2M Client: Test
D: State: 1
D: State: 6
I: RD Client started with endpoint 'Test' with client lifetime 60
D: registration sent [192.168.1.1:4444]
D: Registration callback (code:2.1)
I: *** LWM2M_RD_CLIENT_EVENT_REGISTRATION_COMPLETE
[... 351 lines cut ...]
 - PASS - [lwm2m_rd_client.test_deregistration_timeout] duration = 0.099 seconds
 - FAIL - [lwm2m_rd_client.test_engine_trigger_bootstrap] duration = 1.121 seconds
 - PASS - [lwm2m_rd_client.test_error_on_registration_update] duration = 0.221 seconds
 - PASS - [lwm2m_rd_client.test_network_error_on_registration] duration = 0.142 seconds
 - PASS - [lwm2m_rd_client.test_no_context] duration = 0.012 seconds
 - PASS - [lwm2m_rd_client.test_register_update_too_small_lifetime_to_default] duration = 0.079 seconds
 - PASS - [lwm2m_rd_client.test_registration_update_timeout] duration = 0.236 seconds
 - PASS - [lwm2m_rd_client.test_rx_off] duration = 0.751 seconds
 - PASS - [lwm2m_rd_client.test_socket_error] duration = 0.135 seconds
 - PASS - [lwm2m_rd_client.test_socket_error_on_stop] duration = 1.081 seconds
 - PASS - [lwm2m_rd_client.test_start_registration_fail] duration = 0.072 seconds
 - PASS - [lwm2m_rd_client.test_start_registration_ok] duration = 0.104 seconds
 - PASS - [lwm2m_rd_client.test_start_registration_timeout] duration = 0.065 seconds
 - PASS - [lwm2m_rd_client.test_start_registration_update] duration = 0.215 seconds
 - PASS - [lwm2m_rd_client.test_start_registration_update_fail] duration = 0.252 seconds
 - PASS - [lwm2m_rd_client.test_suspend_resume_registration] duration = 2.910 seconds
 - PASS - [lwm2m_rd_client.test_suspend_stop_resume] duration = 0.089 seconds
 - PASS - [lwm2m_rd_client.test_timeout_resume_registration] duration = 0.098 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED

```

aescolar, 2023-09-01:
@SeppoTakalo an immediate fix may to only allow this test in native_posix.

But I'm getting the feeling this test only passes if some set of events happen in the exact right order which is not at all given, and that the opposite is not sign of a bug. Meaning, that the test relies very particularly on the relative order of events/interrupts in native posix, and therefore may start failing also on native_posix if something happens to change the event order in a way that is still valid (outside of this test).
Is my fear founded?

PR #62283 (2023-09-11): test: lwm2m: Fix minor timing issue
RD client tests work well on emulated (fast) environments but tend to fail on real HW with real time sleeps. This change refactors wait_for_service() to be a bit more syncronous on background service, instead of relying hardcoded sleeps.

Fixes #61824
commit 47981878 (2023-09-11): test: lwm2m: Fix minor timing issue

RD client tests work well on emulated (fast) environments
but tend to fail on real HW with real time sleeps.
This change refactors wait_for_service() to be a bit more
syncronous on background service, instead of relying hardcoded
sleeps.

Fixes #61824
closed 2023-09-11 by carlescufi