zephyrproject-rtos/zephyr #61829: JSON.c obj_parse has wrong datatype

ShahinHaque97-CT, 2023-08-24:
In the file - [https://github.com/zephyrproject-rtos/zephyr/blob/main/lib/os/json.c](url)

static int64_t obj_parse(struct json_obj *obj, const struct json_obj_descr *descr,
			 size_t descr_len, void *val)
```
{
	struct json_obj_key_value kv;
	int64_t decoded_fields = 0;
	size_t i;
        **int ret;**
..
..
```

"int ret" should be changed to "int64_t ret"
The reasoning behind this is, ret returns 32 bits and each member in the struct is error checked by bitwise and so only an array size of 32 can currently be added else an overflow will occur returning a negative. changing it to int64_t allows the full range.

BOARD = MIMXRT1042
SDK = zephyr-sdk-0.16.2-rc1

CONFIG_JSON_LIBRARY=y
#include <zephyr/data/json.h>

ItsRoy69, 2023-08-27:
Can I work on this issue ?

henrikbrixandersen, 2023-08-28:
> Can I work on this issue ?

Patches are always welcome!

ShahinHaque97-CT, 2023-08-28:
> Can I work on this issue ?

Yes of course!, This issue only related to my problem, there may be similar datatype issue in the other functions that I have not checked yet.

ShahinHaque97-CT, 2023-08-30:
I have found another issue which is possibly the same issue in a different location:

Having other datatypes (e.g. uint8_t uint32_t) in the structures returns incorrect values, however int32_t is correct, In most cases, int and int32_t are essentially equivalent.

ShahinHaque97-CT, 2023-10-04:
@ItsRoy69  have noticed this change hasn't been updated in zephyr. I have created a pull request for this
https://github.com/zephyrproject-rtos/zephyr/pull/63499

matt-wood-ct, 2023-10-23:
@mrfuchs Is there anychance someone could do a quick patch for this, it is genuinely this simple:
[image]
But I think myself and Shahin are not well enough versed in the contributor guidelines & processes yet to make this merge go through gracefully.

closed 2024-01-06 by github-actions[bot]