zephyrproject-rtos/zephyr #61834: ESP32-C3 Iterable sections inside ROM doesn't work

horakdavid, 2023-08-24:
**Describe the bug**
When using iterable sections that is located inside ROM, structures are located on wrong address resulting system crash.

I'm using `esp32c3_devkitm` with Zephyr 3.4.

**To Reproduce**
The following sample will crash: 
```
// main.c

#include <zephyr/kernel.h>
#include <zephyr/logging/log.h>

LOG_MODULE_REGISTER(main, LOG_LEVEL_INF);	

struct my_data {
	const char* s;
};

#define DEFINE_DATA(name, _s) \
         STRUCT_SECTION_ITERABLE(my_data, name) = { \
                 .s = _s, \
         }

DEFINE_DATA(d1, "test1");
DEFINE_DATA(d2, "test2");
DEFINE_DATA(d3, "test3");

int main(void)
{
	LOG_INF("start");
	STRUCT_SECTION_FOREACH(my_data, data) {
            LOG_INF("%s", data->s);
	}

	return 0;
}
```

```
# prj.conf
CONFIG_LOG=y
```

```
# CMakeLists.txt
zephyr_linker_sources(SECTIONS iterables.ld)
```

```
#iterables.ld
ITERABLE_SECTION_ROM(my_data, 4)
```

**Expected behavior**
```
*** Booting Zephyr OS build zephyr-v3.4.0-670-g2119df306f5c ***
[00:00:00.000,000] <inf> main: start
[00:00:00.000,000] <inf> main: test1
[00:00:00.000,000] <inf> main: test2
[00:00:00.000,000] <inf> main: test3
```

**Impact**
Some Zephyr's components uses iterable sections (for example mcumgr) and can't be used due to this issue.

**Logs and console output**
Console output when iterable sections is located inside ROM:
```
*** Booting Zephyr OS build zephyr-v3.4.0-670-g2119df306f5c ***
[00:00:00.000,000] <inf> main: start
[00:00:00.000,000] <err> os: 
[00:00:00.000,000] <err> os:  mcause: 5, Load access fault
[00:00:00.000,000] <err> os:   mtval: 432c
[00:00:00.000,000] <err> os:      a0: 00000001    t0: 0000002e
[00:00:00.000,000] <err> os:      a1: 00000002    t1: 420128d8
[00:00:00.000,000] <err> os:      a2: 00000006    t2: ba4c9b13
[00:00:00.000,000] <err> os:      a3: 60023008    t3: 00000008
[00:00:00.000,000] <err> os:      a4: 00000088    t4: 00000004
[00:00:00.000,000] <err> os:      a5: 00000000    t5: 0000004c
[00:00:00.000,000] <err> os:      a6: 00000000    t6: 0000006c
[00:00:00.000,000] <err> os:      a7: 3c000040
[00:00:00.000,000] <err> os:      ra: 420100f6
[00:00:00.000,000] <err> os:    mepc: 4201011c
[00:00:00.000,000] <err> os: mstatus: 00001880
[00:00:00.000,000] <err> os: 
[00:00:00.000,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.000,000] <err> os: Current thread: 0x3fc83970 (unknown)
[00:00:00.097,000] <err> fatal_error: Resetting system
```

**Environment:**
 - OS: Ubuntu 22.04
 - Toolchain: Zephyr SDK 0.16.1
 - Zephyr 3.4

martinjaeger, 2023-09-10:
@sylvioalves do you already have any further insights on this bug? I'm seeing the same issue in a custom application that uses iterable sections.

martinjaeger, 2023-09-12:
This can also be reproduced with the iterable sections test:

```
west build -b xiao_esp32c3 tests/misc/iterable_sections/ -p
west flash
```

Result:

```
*** Booting Zephyr OS build zephyr-v3.4.0-3569-g7b3d94cda20d ***
Running TESTSUITE iterable_sections
===================================================================
START - test_ram
 PASS - test_ram in 0.001 seconds
===================================================================
START - test_rom
E: 
E:  mcause: 5, Load access fault
E:   mtval: 5648
E:      a0: 00000000    t0: 00000088
E:      a1: 3c0020bc    t1: 3fc835f0
E:      a2: 00000000    t2: 40381dae
E:      a3: 00000000    t3: 00000000
E:      a4: 00005658    t4: 00000000
E:      a5: 00005648    t5: 00000000
E:      a6: 00000000    t6: 00000000
E:      a7: 3c0020a8
E:      ra: 42012d3e
E:    mepc: 4201015a
E: mstatus: 00001880
E: 
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x3fc835f0 (test_rom)
E: Halting system
```

PR #62583 (2023-09-18): soc: espressif: provide VMA to rodata and text by default
Flash segments require VMA to proper work. Executing from LMA is not possible. Current implementation did not take into account runtime iterable rom sections that any application could implement. In the above cenario and as reported in the issue below, ESP32 won't run when those ROM sections are created in application level.

This change make sure all flash segments are properly mapped accordingly.

Fixes #61834
commit f5fa4b3b (2023-09-18): soc: espressif: provide VMA to rodata and text by default

Flash segments require VMA to proper work. Executing from LMA
is not possible. Current implementation did not take into account
runtime iterable rom sections that any application could implement.
In the above cenario and as reported in the issue below, ESP32 won't run
when those ROM sections are created in application level.

This change make sure all flash segments are properly mapped
accordingly.

Fixes #61834
commit 3e75c221 (2023-09-19): soc: espressif: provide VMA to rodata and text by default

Flash segments require VMA to proper work. Executing from LMA
is not possible. Current implementation did not take into account
runtime iterable rom sections that any application could implement.
In the above cenario and as reported in the issue below, ESP32 won't run
when those ROM sections are created in application level.

This change make sure all flash segments are properly mapped
accordingly.

Fixes #61834

(cherry picked from commit f5fa4b3bcdd8af30ac2a8d95b32be9776cd6b953)
closed 2023-09-18 by fabiobaltieri