zephyrproject-rtos/zephyr #61863: nrf5340 the CDC_ACM sample does not work suspended/resumed' .  "device descriptor read/64, error -71"

hww, 2023-08-24:
**Describe the bug**

I try to compile a sample of a USB device for nrf5340. No one of them work as it should. For example compiling cdc_acm.

A clear and concise description of what the bug is.

Please also mention any information which could help others to understand
the problem you're facing:
 - Target platform nrf5340
 - Toolschain zephyr-sdk-0.16.1
 - To diagnose issue was done:
    - Enabled Debug level log for USB stack
    - Checked wire connection between USB connector and MCU. 
    -  Tested all another USB samples HID, Bulk -- no one work
    -  Captured the diagrams D+ and D- also connection of VBUS.
- It is this a regression! Same for SDK  v2.4 and  v2.2? 

**To Reproduce**

Steps to reproduce the behavior:

1. copy cdc_acm to ~/
2. open with VS Code
3. edit build settings 
  -  change target nrf5340dk_nrf5340_cpuapp
  -  change configuration to prj.conf
4. Open run menuconfig and change setting:
  - [Max compiled-in log level for usb device] to value (Debug)
5. Build project
6. Flash firmwar

**Expected behavior**

A clear and concise description of what you expected to happen.

Plug USB cable to device shout produce the next ```dmesg```

```sh
   usb 9-1: new full-speed USB device number 112 using uhci_hcd
   usb 9-1: New USB device found, idVendor=8086, idProduct=f8a1
   usb 9-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
   usb 9-1: Product: CDC-ACM
   usb 9-1: Manufacturer: Intel
   usb 9-1: SerialNumber: 00.01
   cdc_acm 9-1:1.0: ttyACM1: USB ACM device
```

But instead it shows

```sh
[ 6192.854409] usb 3-2: new full-speed USB device number 2 using xhci_hcd
[ 6192.994432] usb 3-2: device descriptor read/64, error -71
[ 6193.241087] usb 3-2: device descriptor read/64, error -71
[ 6193.477736] usb 3-2: new full-speed USB device number 3 using xhci_hcd
[ 6193.611085] usb 3-2: device descriptor read/64, error -71
[ 6193.854425] usb 3-2: device descriptor read/64, error -71
[ 6193.961150] usb usb3-port2: attempt power cycle
[ 6194.367734] usb 3-2: new full-speed USB device number 4 using xhci_hcd
[ 6194.367815] usb 3-2: Device not responding to setup address.
[ 6194.575253] usb 3-2: Device not responding to setup address.
[ 6194.784392] usb 3-2: device not accepting address 4, error -71
[ 6194.907727] usb 3-2: new full-speed USB device number 5 using xhci_hcd
[ 6194.908250] usb 3-2: Device not responding to setup address.
[ 6195.114461] usb 3-2: Device not responding to setup address.
[ 6195.321064] usb 3-2: device not accepting address 5, error -71
[ 6195.321142] usb usb3-port2: unable to enumerate USB device
```
The MCU log has the next result

```sh
[... 138 lines cut ...]

....
....
....
[00:00:03.206,298] <inf> usb_cdc_acm: Device suspended

```

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

thedjnK, 2023-08-24:
You're using NCS, bugs for that go to https://devzone.nordicsemi.com/

If you can reproduce the issue with zephyr i.e. not NCS, then the bug should be reported here

jfischer-no, 2023-08-24:
NCS issue or issue caused by the user.

closed 2023-08-24 by jfischer-no