zephyrproject-rtos/zephyr #61866: esp32: spi: clock idle polarity (CPOL) incorrect

MrMarteng, 2023-08-24:
**Describe the bug**
CPOL=0 configurations do not seem to work as intended. Idle state of clock remains high.

**To Reproduce**
All SPI configurations where operation bit `SPI_MODE_CPOL`is not set are effected. 
I create a small test-example for verification
```
struct spi_config config;
struct spi_buf_set spi_bufset;
struct spi_buf spi_buf;
uint8_t data[] = {0xAB, 0x12, 0x00, 0x33, 0xAA};

memset(&config.cs, 0, sizeof(config.cs));
config.frequency = 600000;
config.operation = SPI_OP_MODE_MASTER | SPI_TRANSFER_MSB | SPI_WORD_SET(8) | SPI_MODE_CPHA;

LOG_INF("SPI-OP 0x%x", config.operation);

spi_buf.buf = data;
spi_buf.len = sizeof(data);

spi_bufset.buffers = &spi_buf;
spi_bufset.count = 1;

while (true)
{
    spi_write(spi_dev, &config, &spi_bufset);

    k_sleep(K_SECONDS(2));
}
```

**Expected behavior**
Clock idle state should be low and data line (MOSI) should change on 

**Impact**
SPI Mode0 and Mode1 are not working correctly. Some devices may have difficulties with this.

**Logs and console output**
[image]

**Environment (please complete the following information):**
 - OS: MacOS
 - Board: M5Stack-Core2
 - Commit-ID: 8e7c9a35aebd412c8d432815f50bdac193df337e

**Additional context**
It looks like ESP HAL layer `spi_ll_master_set_mode()` does not receive the actual SPI mode value.

EricNRS, 2023-08-27:
@MrMarteng - have you tried using `line-idle-low` that was added in commit d7bcac091c8?

MrMarteng, 2023-08-28:
> @MrMarteng - have you tried using `line-idle-low` that was added in commit [d7bcac0](https://github.com/zephyrproject-rtos/zephyr/commit/d7bcac091c8c8e99588b25cbe5d95b5360f2f450)?

No I haven’t. But meanwhile I realized that it’s not only the idle polarity. The mode generally is not configured correctly. With my proposed bugging the spi interface is working as intended again.

EricNRS, 2023-08-28:
Yes, the ESP32* SPI peripheral is highly configurable and a lot of options aren't available in other SPI peripheral, so there isn't a config option for it.  There was a lot of discussion that went into the `line-idle-low` debate that may be interesting:  https://github.com/zephyrproject-rtos/zephyr/pull/60183

Granted, most of that is for the spi_write_dt() option and you are using the spi_write() option where you pass the configuration in which is probably entirely different.

MrMarteng, 2023-08-29:
Resolved with #61867

closed 2023-08-29 by MrMarteng