zephyrproject-rtos/zephyr #61873: Nucleo dtsi not such file or directory.

RaysoftTechnologiesInc, 2023-08-25:
Hi,

I am a bit new to Zephyr and I am currently trying to run  a blinky on my nucleo board, wl55jc, However I am currently getting this error
[image]

I have tried running west update but it is not changing anything.

Could you please assist?

henrikbrixandersen, 2023-08-25:
The file in question is provided by the `hal_st` module. Please ensure your manifest contains this module before running `west update`:

```
$ west list hal_st
hal_st       modules/hal/st               9b128caf3e7b2e750169b880e83f210ea2213473 https://github.com/zephyrproject-rtos/hal_st

$ git describe 
zephyr-v3.4.0-2797-g075a859869

$ ls -l ../modules/hal/stm32/dts/st/wl/stm32wl55jcix-pinctrl.dtsi 
-rw-r--r-- 1 hebad hebad 20367 May 10 11:12 ../modules/hal/stm32/dts/st/wl/stm32wl55jcix-pinctrl.dtsi

```

closed 2023-08-25 by henrikbrixandersen

RaysoftTechnologiesInc, 2023-08-25:
Hi,
it seems the module does not exist
[image]

I have been running the west update, isn't it supposed to update everything?

henrikbrixandersen, 2023-08-25:
You are not using upstream Zephyr. You are using the NordicSemi downstream SDK (NCS). It only includes the NordicSemi HAL.

henrikbrixandersen, 2023-08-25:
In the future, please use our bug template when reporting bugs. If you had included the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md this would have been obvious from the start.