zephyrproject-rtos/zephyr #61888: `k_mem_slab_free()` should take `void *mem` instead of `void **mem`

carlescufi, 2023-08-25:
**Describe the bug**

The following function:
```c
void k_mem_slab_free(struct k_mem_slab *slab, void **mem)
```
has the wrong signature. `mem` is only used as a regular pointer, so there is no need to use a double-pointer. The correct signature should be:

```c
void k_mem_slab_free(struct k_mem_slab *slab, void *mem)
```

The issue with the current signature, although functional, is that it is extremely confusing. I myself, a veteran Zephyr developer was confused by this parameter when looking at it.

**Impact**
This impacts the developer in two ways:

1. It requires the developer to unnecessarily compute the address with `&` before calling the function
2. It confuses the user by implying that the pointer itself will be modified

**Additional context**
Note that `k_mem_slab_alloc()` takes a double pointer, but in that case it makes sense, since the result is stored in it.

carlescufi, 2023-08-25:
Options to fix this:

Option a. Use the [Stable API change procedure](https://docs.zephyrproject.org/latest/develop/api/api_lifecycle.html#introducing-incompatible-changes) to introduce the change, would require out-of-tree users to make the change with no deprecation.

Option b. Deprecate the current API, introduce a new one. Perhaps `k_mem_slab_free2()` ?

Option c. Create a new Kconfig option: `CONFIG_K_MEM_SLAB_FREE_OLD_SIGNATURE` that would be enabled by default and would emit a warning (one that does not break the build). After 2 releases remove the Kconfig option.

gmarull, 2023-08-25:
Option (a) please, let’s be practical.

de-nordic, 2023-08-25:
Just a note: there is use for ** in free functions, when the free function is supposed to set invalid address to passed pointer after releasing memory.

carlescufi, 2023-08-25:
> Just a note: there is use for ** in free functions, when the free function is supposed to set invalid address to passed pointer after releasing memory.

Sure, but this is not what the original function was doing, nor I think we should add this functionality, it is counter-intuitive in my opinion.

de-nordic, 2023-08-25:
> > Just a note: there is use for ** in free functions, when the free function is supposed to set invalid address to passed pointer after releasing memory.
> 
> Sure, but this is not what the original function was doing, nor I think we should add this functionality, it is counter-intuitive in my opinion.

Yes.
Sure in world used to malloc/free it is counter-intuitive.

JonathanVanWin, 2023-08-27:
> Just a note: there is use for ** in free functions, when the free function is supposed to set invalid address to passed pointer after releasing memory.

Yeah, I also thought that was the reason it received **.

PR #61901 (2023-09-03): kernel: Modify the signature of `k_mem_slab_free` via stable API change
kernel: Modify the signature of `k_mem_slab_free()` via the [Stable API Change procedure](https://docs.zephyrproject.org/latest/develop/api/api_lifecycle.html#introducing-incompatible-changes)

Modify the signature of the `k_mem_slab_free()` function with a new one,
replacing the old `void **mem` with `void *mem` as a parameter.

Fixes #61888.

Stable API change RFC: #61937

Alternative to https://github.com/zephyrproject-rtos/zephyr/pull/61897
commit 8c748fd0 (2023-09-03): kernel: Modify the signature of k_mem_slab_free()

Modify the signature of the k_mem_slab_free() function with a new one,
replacing the old void **mem with void *mem as a parameter.

The following function:
void k_mem_slab_free(struct k_mem_slab *slab, void **mem);

has the wrong signature. mem is only used as a regular pointer, so there
is no need to use a double-pointer. The correct signature should be:
void k_mem_slab_free(struct k_mem_slab *slab, void *mem);

The issue with the current signature, although functional, is that it is
extremely confusing. I myself, a veteran Zephyr developer,  [... cut]
closed 2023-09-03 by nashif