zephyrproject-rtos/zephyr #61931: NoBufs error when SRP client on OpenThread node registers host and service

koenvervloesem, 2023-08-27:
**Describe the bug**

When trying to register a host and service with the SRP client in the Zephyr shell on an OpenThread node, both stay in a state **ToAdd** and there's a **NoBufs** error message shown by the callback.

This is done with the latest Zephyr main tree, on an nRF52840 Dongle.

**To Reproduce**

I built the echo server example with the OpenThread overlay configuration, adding the configuration option to enable the SRP client:

```
west build -p auto -b nrf52840dongle_nrf52840 samples/net/sockets/echo_server -- -DOVERLAY_CONFIG="overlay-ot.conf" -DCONFIG_OPENTHREAD_SRP_CLIENT=y
```

After flashing the application to the nRF52840 Dongle and connecting to the Thread network, I tried the following to register an SRP host and service:

```
uart:~$ ot srp client host name ot-example
Done
uart:~$ ot ipaddr
fd45:ebce:9ca5:1:c693:db7e:8e3d:dccc
fd3a:3a7a:3ffe:406f:0:ff:fe00:744c
fd3a:3a7a:3ffe:406f:a884:ae04:f665:df2b
fe80:0:0:0:8c26:45bb:27af:ff68
Done
uart:~$ ot srp client host address fd45:ebce:9ca5:1:c693:db7e:8e3d:dccc
Done
uart:~$ ot srp client service add ot-service _example._udp 1234
Done
uart:~$ ot srp client callback enable
Done
uart:~$ ot srp client autostart enable
Done
SRP client callback - error:NoBufs
Host info:
    name:"ot-example", state:ToAdd, addrs:[fd45:ebce:9ca5:1:c693:db7e:8e3d:dcccuart:~$]
Service list:
    instance:"ot-service", name:"_example._udp", state:ToAdd, port:1234, priority:0, weight:0
```

**Expected behavior**

The status of the host and service should be **Registered**, and on the SRP server I should see the host and service.

**Impact**

This makes the SRP client non-functional.

**Environment**

- OS: Ubuntu 22.04 LTS
- Toolchain: Zephyr SDK
- Zephyr Commit SHA: 169f505226a5c9e30732304c6187b0eecd26cfdc

rlubos, 2023-08-28:
Hi @koenvervloesem,

I'm not familiar with that certain OT feature, but a little bit of debugging showed  that SRP client fails at some mbed TLS crypto operation, increasing mbed TLS heap size helped with that particular error in my case (`CONFIG_MBEDTLS_HEAP_SIZE=8192`) - can you check if it's also the case for you?

koenvervloesem, 2023-08-28:
Ok that works, thanks! I actually tried that option, but didn't increase the heap size enough (tried up to 2048).

```
uart:~$ ot srp client autostart enable
Done
SRP client callback - error:OK
Host info:
    name:"ot-example", state:Registered, addrs:[fd45:ebce:9ca5:1:c693:db7e:8e3d:dccc]
Service list:
    instance:"ot-service", name:"_example._udp", state:Registered, port:1234, priority:0, weight:0
```

rlubos, 2023-08-29:
> Ok that works, thanks! 

Can the issue be closed then, since it's not really a bug but rather a misconfiguration issue?

koenvervloesem, 2023-08-29:
Yes, sure, I'll close it, thanks again!

closed 2023-08-29 by koenvervloesem