zephyrproject-rtos/zephyr #61937: RFC: Stable API Change: `k_mem_slab_free()`

carlescufi, 2023-08-27:
### Problem description

See https://github.com/zephyrproject-rtos/zephyr/issues/61888

### Proposed change

See https://github.com/zephyrproject-rtos/zephyr/issues/61888.

## Detailed RFC

See https://github.com/zephyrproject-rtos/zephyr/issues/61888

### Dependencies

None

nashif, 2023-08-28:
> ### Dependencies
> None

Is this really the case?

Per the docs:

   - Dependencies:
             - Impact to users of the API, including the steps required
               to adapt out-of-tree users of the API to the change

So you are saying there is no impact? If I take the first commit from https://github.com/zephyrproject-rtos/zephyr/issues/61888 and run twister on the code base, I get no build errors and no warnings, many samples and tests would fail however. 
Given that we are not deprecating the API, it means that user code will stop working as expected and that is going to be difficult to detect and debug. 
The only information we give the user is a one-liner in the release notes, which is going to be easy to miss.

carlescufi, 2023-08-29:
> Per the docs:
>
> * Dependencies:
>   - Impact to users of the API, including the steps required
>   to adapt out-of-tree users of the API to the change
>
> So you are saying there is no impact? If I take the [...]

No, I clearly misread the docs, thanks for pointing it out. I have updated this now.

carlescufi, 2023-09-04:
Complete

closed 2023-09-04 by carlescufi