zephyrproject-rtos/zephyr #61952: west: build: puncover unknown target

butok, 2023-08-28:
**Describtion**
Followed the "Build Target: puncover" guide: https://docs.zephyrproject.org/latest/develop/optimizations/tools.html#build-target-ram-report
The `west build -t puncovert` command generates an error. It does not see the `puncover` target

**To Reproduce**
Follow instructions provided by: https://docs.zephyrproject.org/latest/develop/optimizations/tools.html#build-target-puncover
1. `pip3 install git+https://github.com/HBehrens/puncover --user`
2. `west build -b reel_board samples/hello_world`
3. `west build -t puncover`
4. See error:
    ```
    C:\_ddm\ZEPHYR\zephyr_latest\zephyrproject\zephyr>west build -t puncover
    -- west build: running target puncover
    ninja: error: unknown target 'puncover'
    FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\_ddm\ZEPHYR\zephyr_latest\zephyrproject\zephyr\build' --target puncover
    ```

**Expected behavior**
To see a report without errors.

**Environment (please complete the following information):**
- Windows 10
- Zephyr SDK 0.16.1
- Zephyr Commit SHA b6095a878c7e9b330dc4a9d91a6a107c544fb47a (22.08.2023)

henrikbrixandersen, 2023-08-29:
Do you have the `puncover` command in your `%PATH`? If not, the CMake build system will not include the `puncover` target.

butok, 2023-08-29:
Strange that the adding path was not required for other python modules.
OK. Added the path. Did not help. Same error -unknown target.

If to run `puncover` as a standalone command, it raising the error: "ImportError: cannot import name 'renderers' from partially initialized module 'puncover'".
```
PS C:\_ddm\ZEPHYR\zephyr_vs\zephyr\samples\basic\blinky> puncover
Traceback (most recent call last):
  File "C:\Users\nxa12205\AppData\Roaming\Python\Python311\site-packages\puncover\puncover.py", line 12, in <module>
    from puncover import renderers
  File "C:\Users\nxa12205\AppData\Roaming\Python\Python311\site-packages\puncover\puncover.py", line 12, in <module>
    from puncover import renderers
ImportError: cannot import name 'renderers' from partially initialized module 'puncover' (most likely due to a circular import) (C:\Users\nxa12205\AppData\Roaming\Python\Python311\site-packages\puncover\puncover.py)
PS C:\_ddm\ZEPHYR\zephyr_vs\zephyr\samples\basic\blinky> 
```

henrikbrixandersen, 2023-08-29:
> Strange that the adding path was not required for other python modules. OK. Added the path. Did not help. Same error -unknown target.

Other Python modules may not install executables.

> If to run `puncover` as a standalone command, it raising the error: "ImportError: cannot import name 'renderers' from partially initialized module 'puncover'".

I would try to get the stand-alone `puncover` command working first, before trying to execute it via Zephyr CMake.

On my system, running `puncover` in shell yields the following:
```
$ puncover
usage: puncover [-h] [--arm_tools_dir ARM_TOOLS_DIR] [--gcc_tools_base GCC_TOOLS_BASE] --elf_file ELF_FILE [--src_root SRC_ROOT] [--build_dir BUILD_DIR] [--debug] [--port PORT] [--host HOST]
puncover: error: the following arguments are required: --elf_file
```

butok, 2023-08-29:
The same issue: https://devzone.nordicsemi.com/f/nordic-q-a/88536/how-to-use-puncover-to-estimate-the-maximum-stack-size-of-a-thread

henrikbrixandersen, 2023-08-29:
> Now see right puncover usage message, but the target build has the same status 1:
> 
> ```
> PS C:\_ddm\ZEPHYR\zephyr_vs\zephyr\samples\basic\blinky> puncover
> usage: puncover [-h] [--gcc-tools-base GCC_TOOLS_BASE] --elf_file ELF_FILE [--src_root SRC_ROOT] [--build_dir BUILD_DIR] [--debug]
>                 [--port PORT] [--host HOST] [--no-open-browser] [--version]
> puncover: error: the following arguments are required: --elf_file/--elf-file
> PS C:\_ddm\ZEPHYR\zephyr_vs\zephyr\samples\basic\blinky> west build -t puncover
> -- west build: running target puncover
> ninja: error: unknown target 'puncover'
> FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\_ddm\ZEPHYR\zephyr_vs\zephyr\samples\basic\blinky\build' --target puncover
> ```

Did you remove the existing build environment and re-run cmake?

> Could you test it on Windows?

I don't have access to a Windows PC.

butok, 2023-08-29:
> Did you remove the existing build environment and re-run cmake?

Removing the build directory, and rebuild helped. Thank you.

But why it helped? Just guessing, the west checks existence of puncover and prepare the target for it. 
Otherwise, it's a magic, as used the same elf file.

henrikbrixandersen, 2023-08-29:
> > Did you remove the existing build environment and re-run cmake?
> 
> Removing the build directory, and rebuild helped. Thank you.
> 
> But why it helped? Just guessing, the west checks existence of puncover and prepare the target for it. Otherwise, it's a magic, as used the same elf file.

The build system is generated by CMake once and the results from looking up available binaries are cached. If you change the environment (e.g. adding a binary which was not present before), you need to regenerate the build system for it to pick up the changes.

closed 2023-08-29 by henrikbrixandersen

butok, 2023-08-29:
Understand. Thank you.