zephyrproject-rtos/zephyr #61961: Build error with Intel SSP/DMIC DMA drivers with check priority Kconfig enabled

kv2019i, 2023-08-28:
**Describe the bug**
SOF builds break after Zephyr commit
```
commit 058f256ea140026e46e458dd51036ea2c1f81b2a
Author: Fabio Baltieri <fabiobaltieri@google.com>
Date:   Fri Mar 10 17:26:35 2023 +0000

    Kconfig.zephyr: check priorities by default 
```

The build error is as follow:
```
20:10:32,457 INFO  - ERROR: /soc/ssp@28000 POST_KERNEL 42 83 < /soc/dfpmccu@71b00/io0_domain POST_KERNEL 75 70
20:10:32,457 INFO  - ERROR: /soc/ssp@29000 POST_KERNEL 42 84 < /soc/dfpmccu@71b00/io0_domain POST_KERNEL 75 70
20:10:32,457 INFO  - ERROR: /soc/ssp@2a000 POST_KERNEL 42 85 < /soc/dfpmccu@71b00/io0_domain POST_KERNEL 75 70
20:10:32,457 INFO  - ERROR: /soc/ssp@2b000 POST_KERNEL 42 86 < /soc/dfpmccu@71b00/io0_domain POST_KERNEL 75 70
20:10:32,457 INFO  - ERROR: /soc/ssp@2c000 POST_KERNEL 42 87 < /soc/dfpmccu@71b00/io0_domain POST_KERNEL 75 70
20:10:32,458 INFO  - ERROR: /soc/ssp@2d000 POST_KERNEL 42 88 < /soc/dfpmccu@71b00/io0_domain POST_KERNEL 75 70
20:10:32,458 INFO  - ERROR: /soc/dmic0@10000 POST_KERNEL 70 74 < /soc/dfpmccu@71b00/hub_ulp_domain POST_KERNEL 75 73
20:10:32,458 INFO  - ERROR: /soc/dmic1@10000 POST_KERNEL 70 75 < /soc/dfpmccu@71b00/hub_ulp_domain POST_KERNEL 75 73
20:10:32,458 INFO  - ERROR: /soc/dma@72c00 POST_KERNEL 40 69 < /soc/dfpmccu@71b00/hst_domain POST_KERNEL 75 65
20:10:32,458 INFO  - ERROR: /soc/dma@7c000 POST_KERNEL 40 82 < /soc/dfpmccu@71b00/hub_ulp_domain POST_KERNEL 75 73
20:10:32,458 INFO  - ERROR: /soc/dma@7d000 POST_KERNEL 40 71 < /soc/dfpmccu@71b00/io0_domain POST_KERNEL 75 70
20:10:32,458 INFO  - ERROR: /soc/dma@7e000 POST_KERNEL 40 72 < /soc/dfpmccu@71b00/io0_domain POST_KERNEL 75 70
20:10:32,458 INFO  - ERROR: /soc/dma@72600 POST_KERNEL 40 67 < /soc/dfpmccu@71b00/hst_domain POST_KERNEL 75 65
20:10:32,458 INFO  - ERROR: /soc/dma@72400 POST_KERNEL 40 66 < /soc/dfpmccu@71b00/hst_domain POST_KERNEL 75 65
20:10:32,458 INFO  - ERROR: /soc/dma@72800 POST_KERNEL 40 68 < /soc/dfpmccu@71b00/hst_domain POST_KERNEL 75 65
```

**Environment (please complete the following information):**
 - https://github.com/thesofproject/sof/actions/runs/5973406935/job/16205619385?pr=8095

fabiobaltieri, 2023-08-28:
@kv2019i apologies for the breakage, I ran the change through the push CI and fixed all that cases but there were inevitably going to be some fallouts only caught by the full run (the one I fixed) or downstream. The fix should be as easy as tweaking the priority of one of the two devices, I suggest lowering the hardcoded one in `drivers/power_domain/power_domain_intel_adsp.c`. LMK if you want me to send a PR for it.

pjdobrowolski, 2023-08-29:
Live would be easier if in PR was instruction about priorities and how they work.

kv2019i, 2023-08-29:
Thanks @fabiobaltieri ! SOF CI tests with Zephyr main look clean again: https://github.com/thesofproject/sof/actions/runs/6007804408/job/16294496014?pr=8114 

Closing the bug.

closed 2023-08-29 by kv2019i

fabiobaltieri, 2023-08-29:
> > Live would be easier if in PR was instruction about priorities and how they work.
> 
> I just found this https://docs.zephyrproject.org/latest/kernel/drivers/index.html#initialization-levels
> 
> @fabiobaltieri is this the best documentation on this topic?

I guess so, if you want more details about that check script specifically there's something in https://docs.zephyrproject.org/latest/build/cmake/index.html#scripts-build-check-init-priorities-py and you can check out the RFC at https://github.com/zephyrproject-rtos/zephyr/pull/55654. There's really not that much to write about though, it's a priority system, lower priority runs first...