zephyrproject-rtos/zephyr #61964: Suggestion: consider making Base64 encoding optional in SMP packages

bogdanm, 2023-08-28:
Hi, we're using the SMP to upload firmware images to the device, as well as uploading and downloading files (some of which are quite big). We have written our own implementation of an SMP client that is able to upload/download files much faster than `mcumgr`, but still I believe that the transfer rate can be improved if we skip the Base64 encoding step (which is not needed on any of the transports that we use). The SMP protocol has a "flags" field that is currently unused, that looks like a good place to add a "Skip Base64" bit. The idea is that if the client sets this bit, the server doesn't Base64 decode the incoming packet and also doesn't Base64 encode the response packet, everything else remains the same. If there's interest in this, I can start looking at what needs to change in order to make this happen.

carlescufi, 2023-08-28:
+1 from me!

nordicjm, 2023-08-29:
It is not as simple as adding a flag to the options field, the SMP header is part of the base-64 encoded content, so you cannot know if you have a base-64 encoded header or not, and this would absolutely break functionality over shell. I wouldn't really consider implementing this unless it can be shown that the base-64 encoding/decoding (on the module side, don't really care about the host side) is shown to cause a big (20%+) decrease in transfer throughput.

nordicjm, 2023-08-29:
Having had more of a thought about it, this solution is not workable, because 0x06 0x09 is used as start delimiter and \n as end delimiter - those can't appear in base64 output so there is no issue, but without base64 they appear everywhere, so this would need to be an entirely new transport.

bogdanm, 2023-08-30:
Well, I guess this settles it then. Please feel free to close this issue, I
can't login to github right now.

nordicjm, 2023-08-30:
It can remain open and it is possible to implement this but it would just need to be a new transport, it would work similar to how bluetooth and UDP work i.e. no header/footer

bogdanm, 2023-08-30:
Sure, but it seems like a lot of work for not a lot of benefit. Although it
might also have advantages in terms of higher transfer speed and/or lower
MTU requirements for BLE. In any case, you're right, no harm in keeping it
open.

JPHutchins, 2024-03-28:
Just chiming in to say that it's worth it for the reduction in complexity alone: no longer a need to encode packets differently for different transports.  Speed improvement should be at most 4/3, the worst case base64 encoding overhead.

nordicjm, 2024-04-02:
> While base64 may be a requirement for the current implementation of the shell transport, I am not convinced that it's an absolute requirement of the shell transport feature.

It is because a 0x0a or 0x0d would be treated as a newline and returns back to command entry mode, the point in the base64 encoding is so that all data is alphanumeric. And also because if there is a shell connected, it would be trying to display binary characters, which is never a good idea.

JPHutchins, 2024-04-02:
Setting aside the Base64 thing for a moment.

The serial envelope also contains a CRC.  This would need to be maintained.  It wasn’t front of mind for me since I mostly use USB CDC ACM but it would be terrifying to do raw serial without that.

In other words, the dream of unifying the transport packet is impossible - serial must be a special case not matter what.

So, then the only argument against Base64 is for this 4/3 speed improvement.  I can state that as it is, the serial transport (over USB) is about 3x faster than BLE: 55KBps vs 16KBps.  It is a very good user experience.  Where would it run into the limits of the OS, the MCU, the flash, etc?  Why does it need to be faster?

nordicjm, 2026-01-06:
> The serial envelope also contains a CRC. This would need to be maintained. It wasn’t front of mind for me since I mostly use USB CDC ACM but it would be terrifying to do raw serial without that.

That is up to users to decide on if they want to risk it or not.
Anyhow, a draft PR with this feature: https://github.com/zephyrproject-rtos/zephyr/pull/101821 definitely not final, will need to be a whole separate class with different ID etc. and will need to actually check differences in transfer speed and build size. If the difference is very little then might just abandon the whole idea