zephyrproject-rtos/zephyr #61979: MTD api: unified method to work with memory technology devices and their partitioning

Laczen, 2023-08-28:
**Is your enhancement proposal related to a problem? Please describe.**

In zephyr multiple memory technology devices are supported. Examples of these devices are `flash` , `eeprom`, `bbram` and even `retained_mem`. These devices each have their own read/write and (optionally) erase methods to work directly with a device. On top of each device a partitioning method is possible (e.g. flash_map for flash devices, retention for retained_mem devices), but this is not the case for all devices.

When the mtd devices are used a direct (or partitioned) method is utilized to create storage solutions on top of them, e.g. in nvs, fcb, littlefs, disks. This results in a storage solution only to be available on some types of mtd.

The missing of a unified method to enable partitioning results in systems creating their own partitioning method (e.g. `retention` or `boot_info`).

**Describe the solution you'd like**

To enable support for all mtd devices in a generic way a layer could be written on top of them that provides support for 3 simple routines:
```
int mtd_read(const struct mtd_info *info, off_t offset, void *data, size_t len);
int mtd_write(const struct mtd_info *info, off_t offset, const void *data, size_t len);
int mtd_erase(const struct mtd_info *info, off_t offset, size_t len);
```
With
```
enum device_type {	FLASH, EEPROM, BBRAM, RETAINED_MEM};
struct mtd_info_cfg = {
	enum device_type mtd_type;
	const struct device *mtd_device;
	const struct mtd_info *mtd_parent; // This is only needed in case subpartitions are wanted
	const off_t off;
	const size_t size;
	const bool read_only;
};
struct mtd_info {
	const struct mtd_info_cfg *cfg;
	const struct mtd_info_data *data; // Unused at the moment.
};
```
The `device_type` is used to allow mapping to the correct mtd routine.

For devices that do not require a `erase` method the `mtd_erase()` function is created as a "compatibility" function for storage solutions that require it. A storage solution that would like to optimize on certain mtd methods could use the device_type to skip the `mtd_erase` function (e.g. a disk storage solution could skip the erase if it is using `retained_mem` as its mtd).

To allow partitioning of a mtd device the "fixed-partition" compatible would be used to partition a mtd device. A `struct mtd_info` would be created for the complete device and also for each partition that is defined on the device. If a created `struct mtd_info` is defined but 
not used its definition would be removed from the binary during compilation.

**Describe alternatives you've considered**
Leave it as is.

Laczen, 2023-08-28:
@de-nordic (flash), @henrikbrixandersen (eeprom), @nordicjm (retained_mem), @yperess (bbram), @jfisher-no (disk), @danieldegrasse (disk).

Laczen, 2023-08-28:
In linux the name mtd is limited to flash devices (nand flash, nor flash, ...). The name `mtd` is badly chosen as the proposal targets other devices as well.

henrikbrixandersen, 2023-08-29:
While I'd like to see generic support for partitioning MTD devices, I don't think we should introduce any "emulators" in order to do so. The term "emulator" should be reserved for emulation, which - at least in my vocabulary - is not something that is used in production firmware.

We could have a proper MTD _subsystem_ with support for the various backends (flash, EEPROM, BBRAM, ...). However, this will not magically solve the issue of being able to use all sorts of different storage _frontends_ on top (settings, filesystems, NVS, ...) since many of these make assumptions about the underlying technology (e.g. assumes that pages needs erasing before being able to write them, which is not true for a number of MTD technologies).

Laczen, 2023-08-29:
@henrikbrixandersen there might be some confusion about the meaning of mtd devices. In zephyr eeprom are also under the mtd umbrella. In linux eeproms are not considered mtd devices. When referring to mtd devices I would be referring to devices that require erase in pages before being written.

If it is accepted that mtd devices would require erase before being written than the mtd subsystem would solve the problem of support for multiple backends, eeprom, bbram,... could receive the erase before write property trough an emulation (couldn't find a better name yet) layer. The name emulation could be avoided by just calling the compatibles `zephyr,eeprom-mtd`, `zephyr,bbram-mtd`.

Laczen, 2023-08-29:
Maybe the layer for eeprom, bbram, retained_mem could be called the "adaptation" layer.

Laczen, 2023-09-29:
I'm stopping my pursuit to try and unify working with different memories as this could be seen alternatives to existing solutions.

closed 2023-09-29 by Laczen