zephyrproject-rtos/zephyr #62047: reqmgmt: Needs Zephyr Project Charter

gregshue, 2023-08-29:
**Is your enhancement proposal related to a problem? Please describe.**
The Zephyr Project Charter states multiple technical and non-technical constraints but the Charter contents are not in the reqmgmt tree.

**Describe the solution you'd like**
I would like to see the Zephyr Project Charter be in the reqmgmt spec tree with each constraint being traceable.

**Describe alternatives you've considered**
None considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

nashif, 2023-08-30:
> I would like to see the Zephyr Project Charter be in the reqmgmt spec tree with each constraint being traceable.

what are those constraints? Can you please list them here so we can assess if those qualify as requirements?

gregshue, 2023-08-30:
> what are those constraints?

Find "shall" in the Zephyr Project charter. There are 45 occurrences.

> Can you please list them here ...

The purpose of the PR is to review the .sdoc file for:

1. accurate capture of the charter text, 
2. the identification of which statements are and are not a requirement, and
3. assignment of unique identifiers to each requirement.
 
It will be more efficient to do that review in the PR than to paste the entire charter text here for review. If a due diligence review shows there are no constraint statements (requirements) in the charter then the PR gets closed.

nashif, 2023-08-30:
> Find "shall" in the Zephyr Project charter. There are 45 occurrences.

Sorry, but this does not make sense. I am not sure how something like "The Project shall be composed of Platinum, Silver and Associate Members." qualify as a requirement or constraint.

gregshue, 2023-08-30:
Ah ... you're beginning the review already.  Let's discuss that individual item in the PR.

During this larger discussion we will resolve what constitutes a requirement/constraint, where it comes from, and how important it is. One of the things we will find is that some constraints from the charter do not contain the verb "shall" (e.g., 6.c, 7.c).

Another thing we will discover is that there are some constraints on what the Zephyr Project produces for contributor engagement (e.g., membership types) vs. what the Zephyr Project produces for reuse by end users (e.g., test suites). Each are part of what is produced by the Zephyr Project.

nashif, 2023-08-30:
> Ah ... you're beginning the review already. Let's discuss that individual item in the PR.

not really, an I have no idea what PR you are referring to.

gregshue, 2023-08-30:
https://github.com/zephyrproject-rtos/reqmgmt/pull/5

nashif, 2024-02-08:
charter is not a requirement.

closed 2024-02-08 by nashif