zephyrproject-rtos/zephyr #62065: I2S on STM32H7

deyro, 2023-08-30:
Currently I2S is not supported on STM32H7 chips. The driver `i2s_ll_stm32.c` does not build for stm32h7 and throws a few linker errors as mentioned in #45955 

**Describe why you are asking for this support?**

I am using a custom board with STM32H753IIK. The i2s2 bus on the board is connected to an audio codec chip. I would like to be able to record and playback audio in **i2s slave mode**. I have modified the i2s driver based on the comments available in #45955. The driver builds successfully but when I run a sample application I am facing the below problem.

For the moment I am trying to only playback audio but `dma_tx_callback` does not get called. I checked this issue #37038 where the problem was not using the PLL clock on a F4 chip but in my case I have enabled the PLL clock also.

I am using Zephyr 3.4.99

Below are the changes made to the `i2s_ll_stm32` driver
```
#if 1  // <---- enabled this
#define DCACHE_INVALIDATE(addr, size) \
...

#define SPI_STM32_DMA_TX	0x01
#define SPI_STM32_DMA_RX	0x02
static inline uint32_t *ll_func_dma_get_reg_addr(SPI_TypeDef *spi, uint32_t location)
{
#if (CONFIG_SOC_SERIES_STM32H7X)
	if (location == SPI_STM32_DMA_TX) {
		/* use direct register location until the LL_SPI_DMA_GetTxRegAddr exists */
		return &(spi->TXDR);
	}
	/* use direct register location until the LL_SPI_DMA_GetRxRegAddr exists */
	return &(spi->RXDR);
#else
	ARG_UNUSED(location);
	return LL_SPI_DMA_GetRegAddr(spi);
#endif /* st_stm32h7_spi */
}

static int rx_stream_start(struct stream *stream, const struct device *dev)
{
...
	ret = start_dma(stream->dev_dma, stream->dma_channel,
			&stream->dma_cfg,
			/*(void *)LL_SPI_DMA_GetRegAddr(cfg->i2s),*/
			ll_func_dma_get_reg_addr(cfg->i2s, SPI_STM32_DMA_RX),
			stream->src_addr_increment, stream->mem_block,
			stream->dst_addr_increment, stream->fifo_threshold,
			stream->cfg.block_size);
...
}

static int tx_stream_start(struct stream *stream, const struct device *dev)
{
...
	ret = start_dma(stream->dev_dma, stream->dma_channel,
			&stream->dma_cfg,
			stream->mem_block, stream->src_addr_increment,
			/*(void *)LL_SPI_DMA_GetRegAddr(cfg->i2s),*/
			ll_func_dma_get_reg_addr(cfg->i2s, SPI_STM32_DMA_TX),
			stream->dst_addr_increment, stream->fifo_threshold,
			stream->cfg.block_size);
...
}

static int rx_stream_start(struct stream *stream, const struct device *dev) and static int tx_stream_start(struct stream *stream, const struct device *dev)
{
...
#if (CONFIG_SOC_SERIES_STM32H7X)
	LL_SPI_EnableIT_UDR(cfg->i2s);
	LL_SPI_EnableIT_OVR(cfg->i2s);
	LL_SPI_EnableIT_FRE(cfg->i2s);
#else
	LL_I2S_EnableIT_ERR(cfg->i2s);
#endif
...
}

static void rx_stream_disable(struct stream *stream, const struct device *dev) and static void tx_stream_disable(struct stream *stream, const struct device *dev)
{
...
#if (CONFIG_SOC_SERIES_STM32H7X)
	LL_SPI_DisableIT_UDR(cfg->i2s);
	LL_SPI_DisableIT_OVR(cfg->i2s);
	LL_SPI_DisableIT_FRE(cfg->i2s);
#else
	LL_I2S_DisableIT_ERR(cfg->i2s);
#endif
...
}
```

Here is my dts

```
/ {
	soc {
		i2s2: i2s2@40003800 {
			compatible = "st,stm32-i2s-custom";
			#address-cells = <1>;
			#size-cells = <0>;
			reg = <0x40003800 0x400>;
			clocks = <&rcc STM32_CLOCK_BUS_APB1 0x00004000>,
					<&rcc STM32_SRC_PLL1_Q SPI123_SEL(0)>;
			interrupts = <36 0>;
			status = "disabled";
		};
	};
};

&pinctrl {
	/omit-if-no-ref/ i2s2_mck_pc6: i2s2_mck_pc6 {
		pinmux = <STM32_PINMUX('C', 6, AF5)>;
		slew-rate = "very-high-speed";
	};

	/omit-if-no-ref/ i2s2_sdi_pi2: i2s2_sdi_pi2 {
		pinmux = <STM32_PINMUX('I', 2, AF5)>;
		slew-rate = "very-high-speed";
	};
 
	/omit-if-no-ref/ i2s2_sdo_pi3: i2s2_sdo_pi3 {
		pinmux = <STM32_PINMUX('I', 3, AF5)>;
		slew-rate = "very-high-speed";
	}; 
};

i2s_rxtx: &i2s2 {
	status = "okay";
	pinctrl-0 = <&i2s2_ck_pi1 &i2s2_mck_pc6 &i2s2_sdi_pi2 &i2s2_sdo_pi3 &i2s2_ws_pb12>;
	pinctrl-names = "default";
	dmas = <&dmamux1 0 40 (STM32_DMA_PERIPH_TX | STM32_DMA_PRIORITY_HIGH)
			&dmamux1 1 39 (STM32_DMA_PERIPH_RX | STM32_DMA_PRIORITY_HIGH)>;
	dma-names = "tx", "rx";
};

```

Below are the sample application codes. It was made by modifying the `zephyr/samples/drivers/i2s/echo` sample. This application along with my codec driver has been tested on `nrf5340dk_nrf5340_cpuapp` board. The application reads audio data from a file and send to the i2s driver.

```
...
// codec init and settings are done here
...
...
    config.word_size = SAMPLE_BIT_WIDTH;
	config.channels = NUMBER_OF_CHANNELS;
	config.format = I2S_FMT_DATA_FORMAT_I2S;
	config.options = I2S_OPT_BIT_CLK_SLAVE | I2S_OPT_FRAME_CLK_SLAVE;
	config.frame_clk_freq = SAMPLE_FREQUENCY;
	config.mem_slab = &mem_slab;
	config.block_size = BLOCK_SIZE;
	config.timeout = TIMEOUT;
	if (!configure_streams(i2s_dev_rx, i2s_dev_tx, &config)) {
		return -1;
	}
...
...
// Loop
				size_t block_size = BLOCK_SIZE;
				void *buf = calloc(1, block_size);
				if (buf == NULL) {
					LOG_ERR("calloc failed ad %d bytes!!", block_size);
					break;
				}
				rc = fs_read(&fi->file, buf, block_size);
				if (rc <= 0) {
					free(buf);
					LOG_INF("no more data in file");
					break;	
				}

//				ret = i2s_buf_write(i2s_dev_tx, buf, block_size);
				ret = i2s_write(i2s_dev_tx, buf, block_size);
				if (ret < 0) {
					free(buf);
					LOG_ERR("Failed to write data: %d\n", ret);
					err = 1;
					break;
				}

				free(buf);
...
```

My requests are:
1. Can stm32h7 support be added to the i2s driver? I have limited experience in the lower layer drivers.
2. Can someone tell me what am I doing wrong and what else I should be trying?

erwango, 2023-08-30:
Hi,

We have no short term plan to extend I2S support, but here are some hints:
- Did you enabled dmamux and dma nodes in dts ? See [SPI example](https://github.com/zephyrproject-rtos/zephyr/blame/main/tests/drivers/spi/spi_loopback/boards/nucleo_h743zi.overlay#L23)
- Have a check on SPI driver history ([spi_ll_stm32.c](https://github.com/zephyrproject-rtos/zephyr/commits/main/drivers/spi/spi_ll_stm32.c), [spi_ll_stm32.h](https://github.com/zephyrproject-rtos/zephyr/commits/main/drivers/spi/spi_ll_stm32.h))
- Check dts pll configuration. You're using PLL1_Q as clock input. You need to be sure it is correctly configured vs your expectations.
- Ask @endian-benjamin. You already did it on #45955 but let's continue the discussion here if he jumps in.

Good luck

deyro, 2023-08-31:
Hi @erwango,

- Yes, the dmamux and dma nodes are enabled
```
&dma1 {
	status = "okay";
};
&dma2 {
	status = "okay";
};
&dmamux1 {
	status = "okay";
};
```
- Below is the PLL configuration. I believe it is correct
```
&clk_hse {
	hse-bypass;
	clock-frequency = <DT_FREQ_M(24)>; /* 24MHz clock */
	status = "okay";
};
&pll {
	div-m = <2>;
	mul-n = <80>;
	div-p = <2>;
	div-q = <16>;
	div-r = <2>;
	clocks = <&clk_hse>;
	status = "okay";
};
&rcc {
	clocks = <&pll>;
	clock-frequency = <DT_FREQ_M(480)>;
	d1cpre = <1>;
	hpre = <2>;
	d1ppre = <2>;
	d2ppre1 = <2>;
	d2ppre2 = <2>;
	d3ppre = <2>;
};
```

I tried running the `spi_loopback` sample using nucleo_h753zi dts on my custom board, but got the error code -14. The `buf_in_nocache` function returns false. Any idea what is the problem?

Thanks

erwango, 2023-08-31:
> I tried running the spi_loopback sample using nucleo_h753zi dts on my custom board, but got the error code -14. The buf_in_nocache function returns false. Any idea what is the problem?

You should enable `CONFIG_NOCACHE_MEMORY=y`