zephyrproject-rtos/zephyr #62067: tests-ci : drivers: gpio: get_direction test Device issue (Flash error?)

hakehuang, 2023-08-30:
**Describe the bug**
    get_direction test is Device issue (Flash error?) on zephyr-v3.4.0-2851-g34fddd8b4c09 on lpcxpresso55s28
    testcase path is drivers/gpio/get_direction/input_output

    see logs for details

    **To Reproduce**
    1. 
    ```
    scripts/twister --device-testing --device-serial /dev/ttyACM0 -p lpcxpresso55s28      --sub-test drivers.gpio
    ```
    or
    ```
    # cd tests/drivers/gpio/get_direction/input_output
    # west build -b lpcxpresso55s28
    # west flash
    ```
    2. See error

    **Expected behavior**
    test pass

    **Impact**

    **Logs and console output**
    
    ```
    None
        
    ```

    **Environment (please complete the following information):**
     - OS: (e.g. Linux )
     - Toolchain (e.g Zephyr SDK)
     - Commit SHA or Version used: zephyr-v3.4.0-2851-g34fddd8b4c09

nashif, 2023-09-01:
@hakehuang  please fix whatever scripts you are using to create issues, the layout of the bug is malformed.

hakehuang, 2023-09-03:
> @hakehuang please fix whatever scripts you are using to create issues, the layout of the bug is malformed.

sorry, let me check my scripts.

hakehuang, 2023-09-03:
fixed in : 69643d575d6b95b29f2847ebfeaa150d758bb87f

closed 2023-09-03 by hakehuang