zephyrproject-rtos/zephyr #62127: zassert log messages should print values, not variable names

GrantRolls, 2023-08-31:
**Is your enhancement proposal related to a problem? Please describe.**

`zassert_xxx` functions do not print the values being compared, rather it prints the names of the variables. 
This is limiting when using zassert for unit testing in a TDD fashion - immediate feedback on failures is important. 

Currently, to find out what went wrong I have to add custom message to each zassert, but this ends up being a lot of redundant work just to print values.

I think it is less helpful to print the variable names - because you have the line reference in the log

**Describe the solution you'd like**

By default, I think `zassert_xxx` should print the VALUES being compared rather then the name of the variable/function.

For `zassert_mem_equal`, it should print the index of the first mismatch and the value/s at that index

nashif, 2024-05-23:
macros already do that, ie..

```
START - test_rules_before_after

    Assertion failed at WEST_TOPDIR/zephyr/tests/ztest/base/src/main.c:118: rule_after_each: (fixture->state not equal to 1111)
```

closed 2024-05-23 by nashif

GrantRolls, 2024-05-23:
> macros already do that, ie..
> 
> ```
> START - test_rules_before_after
> 
>     Assertion failed at WEST_TOPDIR/zephyr/tests/ztest/base/src/main.c:118: rule_after_each: (fixture->state not equal to 1111)
> ```

In this example, fixture->state should instead print as the value rather then stringify the name

rettichschnidi, 2024-10-04:
@GrantRolls Is #79398 something you'd be using?

GrantRolls, 2024-10-09:
> @GrantRolls Is #79398 something you'd use?

Yeah, this is what I was expecting for a zassert_mem_equal

rettichschnidi, 2024-10-22:
> I think there is room for extension macros for each specific type, ...

Definitely room for improvement. Will try once the current PR gets approved.

> to do a similar value print `zassert_equal_int32`, `zassert_equal_uint8` etc.

Agree, but I'd rather use some macro/generic programming rather than having to spell out the integer types.