zephyrproject-rtos/zephyr #62130: Setup override after setting up sysclock

maxvankessel, 2023-08-31:
According to the reference manual the overdrive can be enabled when the sysclock source is set and  the PLL has been turned on:

[image]

I believe the following code block (including the FLASH ACR LATENCY) should be moved below `set_up_plls();`

https://github.com/zephyrproject-rtos/zephyr/blob/64e841152737507c62f9daa8009be0a0dbfcdc99/drivers/clock_control/clock_stm32_ll_common.c#L726-L739

I'm having issues in combination with MCUboot, standalone I don't have any issues.

nordicjm, 2023-08-31:
We have a template for reporting issues, can you fill it in instead of clearing it and making it impossible to know what you're using?

maxvankessel, 2023-08-31:
> We have a template for reporting issues, can you fill it in instead of clearing it and making it impossible to know what you're using?

Just used the github interface:
[image]

nordicjm, 2023-08-31:
https://github.com/zephyrproject-rtos/zephyr/issues/new/choose is where issues should be raised

@nashif can this other way of raising issues be blocked?

erwango, 2023-08-31:
> I believe the following code block (including the FLASH ACR LATENCY) should be moved below set_up_plls();

Did you have a try?

maxvankessel, 2023-08-31:
> > I believe the following code block (including the FLASH ACR LATENCY) should be moved below set_up_plls();
> 
> Did you have a try?

Yes, but it's not exactly as the reference manual states.

maxvankessel, 2023-08-31:
Do you want me to create a PR @erwango?

closed 2023-09-13 by carlescufi