zephyrproject-rtos/zephyr #62164: Possible mis-alignment between enhanced and not enhanced ATT PDU alloc/create

erbr-ot, 2023-08-31:
After merging recent zephyr main branch into local repo, I notice failure re. PDU lengths in ATT.

I deduce that the reason for the discrepancy in my tests is the IS_DEFINED(CONFIG_BT_EATT) guarded net_buf_reserve() in bt_att_chan_create_pdu(), that reserves the 'extra' two bytes for the K-frame. Removing it and my tests pass again. 
This was added by this commit:
https://github.com/zephyrproject-rtos/zephyr/pull/60309/commits/4d93c0e650fcf58970e95c3e5241e089c2e3e108

Then I wonder, should this guard not be bt_att_is_enhanced(chan) instead, such that any channels that are not enhanced do not reserve the extra bytes?

henrikbrixandersen, 2023-08-31:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

jori-nordic, 2023-09-18:
I fail to reproduce the problem here: 
https://github.com/jori-nordic/zephyr/blob/8629d71f1aec6abe364fc54a200cfb12bdfff1a0/tests/bsim/bluetooth/host/att/pdu/dut/prj.conf#L12

You can run the test like so 
```bash
cd zephyrproject/zephyr/tests/bsim/bluetooth/host/att/pdu
./scripts/_compile.sh && ./scripts/att_pdu.sh 
```
It prints out the filenames of the pcaps, so you can verify the packet lengths. 

Enabling/disabling EATT on the DUT doesn't change anything, even though we _are_ reserving the extra two bytes (I added a LOG_ERR there just in case).

erbr-ot, 2023-09-26:
As it turns out there seems to be some kind of confusion (at least in my head!) regarding MTU configs vs. buffer (pool) sizes vs. RX/TX vs. EATT/ATT 
I see in this work that you use a CONFIG_BT_L2CAP_TX_MTU of 251 - where 'we' use CONFIG_BT_L2CAP_TX_MTU=247. 
If I use a 251 'things' look independant of the 'extra reserve' that I talk about in this issue. 
I some how still feel that it seems off to have a config dependancy here and not a run time dependancy per channel

jori-nordic, 2023-09-26:
I still don't see anything different with `CONFIG_BT_L2CAP_TX_MTU=247`.
With both EATT on and off, it results in a write of 240 bytes of payload.

erbr-ot, 2023-10-10:
I'm sorry for ghosting you @jori-nordic - I'm quite busy hunting some bugs, so I have to not focus on this at the moment.

closed 2023-12-24 by github-actions[bot]