zephyrproject-rtos/zephyr #62172: Fatal error when using MCUboot and Zephyr v3.4.0 on esp32c3

abdulayubzoi, 2023-08-31:
**Describe the bug**
I'm currently working on a BLE peripheral running on an `esp32c3`. I'm using `MCUboot` as the bootloader, and since I need both secure-boot and encryption, I build it from the Espressif port (based on ESP-IDF 5.1.0).

Ever since I upgraded to Zephyr `v3.4.0` from `v3.1.0`, I notice sporadic fatal errors occurring on boot, mostly of the type "Illegal instruction". This does not happen if I use the default Espressif bootloader i.e. the application runs correctly.

I have a feeling this has something to do with the settings subsystem as well, but I'm unable to narrow down the issue to anything specific other than the bootloader.

**To Reproduce**
Although I have a custom application that does not reproduce the error in a deterministic way, I have been able to reproduce the behavior using a sample project, using the following steps:

- Build MCUboot for `esp32c3` using this [guide](https://github.com/mcu-tools/mcuboot/blob/main/docs/readme-espressif.md). Use ESP-IDF `v5.1.0`. Use the default [bootloader.conf](https://github.com/mcu-tools/mcuboot/blob/main/boot/espressif/port/esp32c3/bootloader.conf) for `esp32c3`. Do not enable secure-boot or encryption for the sake of simplicity.
- Take the [peripheral](https://github.com/zephyrproject-rtos/zephyr/tree/zephyr-v3.4.0/samples/bluetooth/peripheral) sample.
- Add `CONFIG_BOOTLOADER_MCUBOOT=y` to the `prj.conf` of the sample.
- Build the application for `esp32c3_devkitm` and flash it to the board along with the bootloader built before.
- Observe the debug logs.

**Expected behavior**
No fatal error on boot.

**Impact**
Showstopper, since I'm unable to run my application properly.

**Logs and console output**

With MCUboot as bootloader (fatal error):
```
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C3
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd85a8,len:0x14a8
load:0x403c7000,len:0x21b0
load:0x403d0000,len:0x2240
entry 0x403d21f0
[esp32c3] [INF] *** Booting MCUboot build v1.10.0-137-g4ff95da6 ***
[esp32c3] [INF] [boot] chip revision: v0.3
[esp32c3] [INF] [boot.esp32c3] SPI Speed      : 40MHz
[esp32c3] [INF] [boot.esp32c3] SPI Mode       : DIO
[esp32c3] [INF] [boot.esp32c3] SPI Flash Size : 4MB
[esp32c3] [INF] [boot] Enabling RNG early entropy source...
[esp32c3] [INF] Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32c3] [INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32c3] [INF] Boot source: primary slot
[esp32c3] [INF] Image index: 0, Swap type: none
[esp32c3] [INF] Disabling RNG early entropy source...
[esp32c3] [INF] br_image_off = 0x10000
[esp32c3] [INF] ih_hdr_size = 0x20
[esp32c3] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32c3] [INF] DRAM segment: start=0x1f34c, size=0xfd0, vaddr=0x3fc99130
[esp32c3] [INF] IRAM segment: start=0x1613c, size=0x9210, vaddr=0x40380000
0x40380000: _esp32c3_vector_table at ??:?

[esp32c3] [INF] start=0x40388f0a
0x40388f0a: __start at /workdir/zephyr/soc/riscv/esp32c3/loader.c:83

[esp32c3] [INF] DROM segment: paddr=00010040h, vaddr=3C000040h, size=060FAh ( 24826) map
[esp32c3] [INF] IROM segment: paddr=00030000h, vaddr=42020000h, size=2A8F4h (174324) map
*** Booting Zephyr OS build zephyr-v3.4.0 ***
[00:00:00.017,000] <inf> fs_nvs: 6 Sectors of 4096 bytes
[00:00:00.017,000] <inf> fs_nvs: alloc wra: 0, f88
[00:00:00.017,000] <inf> fs_nvs: data wra: 0, 7c
[00:00:00.018,000] <inf> esp32_bt_adapter: BT controller compile version [d913766]
[00:00:00.499,000] <inf> bt_hci_core: No ID address. App must call settings_load()
Bluetooth initialized
[00:00:00.501,000] <err> os: 
[00:00:00.501,000] <err> os:  mcause: 2, Illegal instruction
[00:00:00.501,000] <err> os:   mtval: 0
[00:00:00.501,000] <err> os:      a0: 00000000    t0: 0000002e
[00:00:00.501,000] <err> os:      a1: 00003c00    t1: 400510a0
[00:00:00.501,000] <err> os:      a2: 3fc8dbac    t2: e7fd2b22
[00:00:00.501,000] <err> os:      a3: 0000002f    t3: 60002058
[00:00:00.501,000] <err> os:      a4: 0000003d    t4: 00000004
[00:00:00.501,000] <err> os:      a5: 00004202    t5: 0000004c
[00:00:00.501,000] <err> os:      a6: 0000005a    t6: 0000006c
[00:00:00.501,000] <err> os:      a7: 67ff0022
[00:00:00.501,000] <err> os:      ra: 4202342e
[00:00:00.501,000] <err> os:    mepc: 00004202
[00:00:00.501,000] <err> os: mstatus: 00001880
[00:00:00.501,000] <err> os: 
[00:00:00.501,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.501,000] <err> os: Current thread: 0x3fc89f58 (unknown)
[00:00:00.627,000] <err> os: Halting system
```
With default Espressif bootloader (normal behavior):
```
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C3
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x15e8
load:0x403ce000,len:0x920
load:0x403d0000,len:0x2c54
SHA-256 comparison failed:
Calculated: b642c45636f27168ae762f9606c8a904ea8a514576223bdd371e4534dcfdaf07
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000
I (49) boot: ESP-IDF abe2993 2nd stage bootloader
I (49) boot: compile time 14:02:53
I (49) boot: chip revision: 3
I (51) boot.esp32c3: SPI Speed      : 40MHz
I (56) boot.esp32c3: SPI Mode       : DIO
I (60) boot.esp32c3: SPI Flash Size : 4MB
I (65) boot: Enabling RNG early entropy source...
I (70) boot: Partition Table:
I (74) boot: ## Label            Usage          Type ST Offset   Length
I (81) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (89) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (96) boot:  2 factory          factory app      00 00 00010000 00100000
I (104) boot: End of partition table
I (108) esp_image: segment 0: paddr=00010020 vaddr=00000020 size=0001ch (    28) 
I (116) esp_image: segment 1: paddr=00010044 vaddr=3fc98f50 size=00a2ch (  2604) load
I (125) esp_image: segment 2: paddr=00010a78 vaddr=3fc9997c size=005a4h (  1444) load
I (134) esp_image: segment 3: paddr=00011024 vaddr=40380000 size=09030h ( 36912) load
I (151) esp_image: segment 4: paddr=0001a05c vaddr=00000000 size=05fdch ( 24540) 
I (157) esp_image: segment 5: paddr=00020040 vaddr=3c000040 size=06038h ( 24632) map
I (164) esp_image: segment 6: paddr=00026080 vaddr=00000000 size=09f98h ( 40856) 
I (175) esp_image: segment 7: paddr=00030020 vaddr=42010020 size=2a330h (172848) map
I (216) boot: Loaded app from partition at offset 0x10000
I (216) boot: Disabling RNG early entropy source...
*** Booting Zephyr OS build zephyr-v3.4.0 ***
[00:00:00.017,000] <inf> fs_nvs: 6 Sectors of 4096 bytes
[00:00:00.017,000] <inf> fs_nvs: alloc wra: 0, f88
[00:00:00.017,000] <inf> fs_nvs: data wra: 0, 7c
[00:00:00.018,000] <inf> esp32_bt_adapter: BT controller compile version [d913766]
[00:00:00.476,000] <inf> bt_hci_core: No ID address. App must call settings_load()
Bluetooth initialized
[00:00:00.480,000] <inf> bt_hci_core: Identity: C4:C5:6D:0B:90:3E (random)
[00:00:00.480,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0016, manufacturer 0x0060
[00:00:00.480,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x0016
Advertising successfully started
Indicate VND attr 0x3c000488 (UUID 12345678-1234-5678-1234-56789abcdef1)
```

**Environment:**
- Devcontainer based on image: zephyrprojectrtos/zephyr-build:v0.26.2
- Toolchain: Zephyr SDK 0.16.0
- Zephyr branch: v3.4.0
- Target: esp32c3 (DevkitM-1)
- Bootloader: MCUboot (latest or v1.10.0)
- ESP-IDF v5.1.0

nordicjm, 2023-09-01:
Can you use `addr2line` on your output elf file for address `0x3fc89f58`?

abdulayubzoi, 2023-09-01:
> Can you use `addr2line` on your output elf file for address `0x3fc89f58`?

Sure, I used:
`/opt/toolchains/zephyr-sdk-0.16.0/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-addr2line -e build/zephyr/zephyr.elf -a 0x3fc89f58`

Output:
```
0x3fc89f58
/workdir/zephyr/kernel/init.c:53
```
https://github.com/zephyrproject-rtos/zephyr/blob/356c8cbe63ae01b3ab438382639d25bb418a0213/kernel/init.c#L53

I'm not sure how this line could be the issue though.
Note: I've used `west build -b esp32c3_devkitm` to build this sample.

abdulayubzoi, 2023-09-01:
@nordicjm I used `addr2line` on the ELF file for my own project as well, where the address in the error log was `0x3fc8f000`.
I have got the same line as above i.e. `/workdir/zephyr/kernel/init.c:53`.

almir-okato, 2023-09-01:
Hi @abdulayubzoi, I've tested with MCUboot and Zephyr latest `main` revision and it worked as expected:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd85a8,len:0x14cc
load:0x403c7000,len:0x21b0
load:0x403d0000,len:0x2198
entry 0x403d214a
[esp32c3] [INF] *** Booting MCUboot build v1.10.0-141-ge4650b75 ***
[esp32c3] [INF] [boot] chip revision: v0.3
[esp32c3] [INF] [boot.esp32c3] SPI Speed      : 40MHz
[esp32c3] [INF] [boot.esp32c3] SPI Mode       : DIO
[esp32c3] [INF] [boot.esp32c3] SPI Flash Size : 4MB
[esp32c3] [INF] [boot] Enabling RNG early entropy source...
[esp32c3] [INF] Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x1
[esp32c3] [INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32c3] [INF] Boot source: primary slot
[esp32c3] [INF] Image index: 0, Swap type: none
[esp32c3] [INF] Disabling RNG early entropy source...
[esp32c3] [INF] br_image_off = 0x10000
[esp32c3] [INF] ih_hdr_size = 0x20
[esp32c3] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32c3] [INF] DRAM segment: start=0x1f428, size=0xfe0, vaddr=0x3fc99168
[esp32c3] [INF] IRAM segment: start=0x16214, size=0x9214, vaddr=0x40380000
[esp32c3] [INF] start=0x40388f10
[esp32c3] [INF] DROM segment: paddr=00010040h, vaddr=3C000040h, size=061D4h ( 25044) map
[esp32c3] [INF] IROM segment: paddr=00030000h, vaddr=42020000h, size=2ACEEh (175342) map

*** Booting Zephyr OS build zephyr-v3.4.0-3204-ga7b0ce614a2d ***
[00:00:00.018,000] <inf> fs_nvs: 6 Sectors of 4096 bytes
[00:00:00.018,000] <inf> fs_nvs: alloc wra: 0, f88
[00:00:00.018,000] <inf> fs_nvs: data wra: 0, 7c
[00:00:00.019,000] <inf> esp32_bt_adapter: BT controller compile version [d913766]
[00:00:00.474,000] <inf> bt_hci_core: No ID address. App must call settings_load()
Bluetooth initialized
[00:00:00.478,000] <inf> bt_hci_core: Identity: C7:F7:D4:81:1E:88 (random)
[00:00:00.478,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0016, manufacturer 0x0060
[00:00:00.478,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x0016
Advertising successfully started
Indicate VND attr 0x3c000488 (UUID 12345678-1234-5678-1234-56789abcdef1)
``` 

Could you please retry with the latest revisions?

abdulayubzoi, 2023-09-04:
> Hi @abdulayubzoi, I've tested with MCUboot and Zephyr latest `main` revision and it worked as expected:
>
> ```
> ESP-ROM:esp32c3-api1-20210207
> Build:Feb  7 2021
> rst:0x1 (POWERON),boot:0xc [...]

Hi @almir-okato,
I have tested with the latest `main` of zephyr, and it works as you showed. However the `main` branch is not stable enough for my project (too many other issues that I don't see with `v3.4.0`), so I would prefer to stick to `v3.4.0`.

Do you know what causes this bug in `v3.4.0`? Are there any plans to backport the fix?

abdulayubzoi, 2023-09-04:
Hi @almir-okato 

I also tested my own project with the latest `main`, and while I no longer see the issue during normal boots, I face the same issue during a FOTA update. 

I can replicate the issue every time a FOTA update has been transferred to the secondary slot and the module reboots, regardless of the fact that the swapping of images happens or not. For context, I use the following functions to perform an update: 
- `flash_img_init()` to initialize the update context
- `flash_img_buffered_write()` to write the contents of the update image to the secondary slot
- `boot_write_img_confirmed()` to mark the image for update
- `sys_reboot(SYS_REBOOT_WARM)` to reboot

Please let me know if I can share anything else to help narrow down the issue.

abdulayubzoi, 2023-10-05:
I believe this issue is somehow related to this one:
https://github.com/zephyrproject-rtos/zephyr/issues/63062

INemesisI, 2023-11-17:
The commit https://github.com/zephyrproject-rtos/zephyr/pull/63817 mentioned in https://github.com/zephyrproject-rtos/zephyr/issues/63062 does not seem to fix the issue. I tested the latest zephyr master as well.
Not being able to boot at all using mcuboot is a big show-stopper for me at the moment. I would be glad if someone can look into this.

sylvioalves, 2023-11-17:
@INemesisI, can you confirm again the scenario and sample code you are working with?

sylvioalves, 2023-11-17:
I've just tried using latest main:  
MCUBoot build and flash:  
```c
rm -rf build; west build -b esp32c3_devkitm ../bootloader/mcuboot/boot/zephyr/
west flash
```

Sample build and flash:  
```c
rm -rf build; west build -b esp32c3_devkitm samples/bluetooth/periodic_adv -DCONFIG_BOOTLOADER_MCUBOOT=y
west flash; west espressif monitor
```
I got it all working:  
```c
[esp32c3] [INF] MCUboot 2nd stage bootloader
[esp32c3] [INF] compiled at Nov 17 2023 08:16:12
[esp32c3] [INF] Chip revision: 3
[esp32c3] [INF] SPI Flash speed: 40MHz, mode: DIO, size: 4MB
[esp32c3] [INF] Image index: 0, Swap type: none
[esp32c3] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32c3] [INF] Application start=403873E6h
[esp32c3] [INF] DRAM segment: paddr=0000C2A0h, vaddr=3FC96658h, size=00890h (  2192) load
[esp32c3] [INF] IRAM segment: paddr=00004308h, vaddr=40380000h, size=07F98h ( 32664) load
[esp32c3] [INF] DROM segment: paddr=00010040h, vaddr=3C000040h, size=042C8h ( 17096) map
[esp32c3] [INF] IROM segment: paddr=00020000h, vaddr=42020000h, size=1B28Ch (111244) map
*** Booting Zephyr OS build zephyr-v3.5.0-1318-g12e8490b8f0a ***
Starting Periodic Advertising Demo
[00:00:00.001,000] <inf> esp32_bt_adapter: BT controller compile version [d913766]
[00:00:00.166,000] <inf> bt_hci_core: Identity: E7:2D:54:62:82:6F (random)
[00:00:00.166,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0016, manufacturer 0x0060
[00:00:00.166,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x0016
Start Extended Advertising...done.
```

Is the issue related to v3.4.0 version?

closed 2024-02-01 by github-actions[bot]