zephyrproject-rtos/zephyr #62202: Twister --coverage not working with lcov >= 2.0

Thalley, 2023-09-01:
**Describe the bug**
When running `twister --coverage` with `lcov` version 2.0 or newer, no code coverage report is generated. 

There are 2 issues:

1. The `lcov_branch_coverage` argument has been deprecated
2. No code coverage is generate at all, even when `lcov_branch_coverage` has been replaced with `branch_coverage`

**To Reproduce**
Steps to reproduce the behavior:
1. Following https://docs.zephyrproject.org/latest/develop/test/coverage.html#coverage-reports-using-twister running `$ twister --coverage -p qemu_x86 -T tests/kernel` (or any of the other commands examples) will fail when using `lcov` version 2.0 or newer.

**Expected behavior**
Expected a code coverage report to be generate when using `lcov` 2.0 or newer. Currently only works with older versions like 1.5 or 1.6. 

**Impact**
Makes it impossible to run twister with code coverage on Arch Linux or other distros that have a newer version of `lcov` than 1.6, without explicitly downgrading the `lcov` package. 

**Logs and console output**
```
[thalley@thalley-desktop-linux zephyr]$ ./scripts/twister --coverage -T tests/bluetooth/audio/ascs
ZEPHYR_BASE unset, using "/home/thalley/dev/nordic/zephyrproject/zephyr"
Renaming output directory to /home/thalley/dev/nordic/zephyrproject/zephyr/twister-out.15
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.4.0-2936-gbb9c84d41c2e
INFO    - Using 'zephyr' toolchain.
INFO    - Selecting default platforms per test case
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/thalley/dev/nordic/zephyrproject/zephyr/twister-out/testplan.json
INFO    - JOBS: 16
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    5/   5  100%  skipped:    0, failed:    0, error:    0
INFO    - 5 test scenarios (5 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 5 of 5 test configurations passed (100.00%), 0 failed, 0 errored, 0 skipped with 0 warnings in 4.48 seconds
INFO    - In total 556 test cases were executed, 94 skipped on 1 out of total 607 platforms (0.16%)
INFO    - 5 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Generating coverage files...
geninfo: WARNING: RC option 'lcov_branch_coverage' is deprecated.  Consider using 'branch_coverage. instead.  (Backward-compatible support will be removed in the future
geninfo: ERROR: no .gcda files found in /home/thalley/dev/nordic/zephyrproject/zephyr/twister-out
        (use "geninfo --ignore-errors empty ..." to bypass this error)
lcov: WARNING: RC option 'lcov_branch_coverage' is deprecated.  Consider using 'branch_coverage. instead.  (Backward-compatible support will be removed in the future
lcov: ERROR: no valid records found in tracefile /home/thalley/dev/nordic/zephyrproject/zephyr/twister-out/coverage.info
        (use "lcov --ignore-errors empty ..." to bypass this error)
lcov: WARNING: RC option 'lcov_branch_coverage' is deprecated.  Consider using 'branch_coverage. instead.  (Backward-compatible support will be removed in the future
lcov: ERROR: no valid records found in tracefile /home/thalley/dev/nordic/zephyrproject/zephyr/twister-out/coverage.info
        (use "lcov --ignore-errors empty ..." to bypass this error)
lcov: WARNING: RC option 'lcov_branch_coverage' is deprecated.  Consider using 'branch_coverage. instead.  (Backward-compatible support will be removed in the future
lcov: ERROR: no valid records found in tracefile /home/thalley/dev/nordic/zephyrproject/zephyr/twister-out/coverage.info
        (use "lcov --ignore-errors empty ..." to bypass this error)
lcov: WARNING: RC option 'lcov_branch_coverage' is deprecated.  Consider using 'branch_coverage. instead.  (Backward-compatible support will be removed in the future
lcov: ERROR: no valid records found in tracefile /home/thalley/dev/nordic/zephyrproject/zephyr/twister-out/coverage.info
        (use "lcov --ignore-errors empty ..." to bypass this error)
genhtml: ERROR: no valid records found in tracefile /home/thalley/dev/nordic/zephyrproject/zephyr/twister-out/coverage.info
        (use "genhtml --ignore-errors empty ..." to bypass this error)
INFO    - Saving reports...
INFO    - Writing JSON report /home/thalley/dev/nordic/zephyrproject/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/thalley/dev/nordic/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/thalley/dev/nordic/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
[thalley@thalley-desktop-linux zephyr]$ echo $ZEPHYR_TOOLCHAIN_VARIANT
zephyr
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Arch Linux with lcov version 2.0
 - Toolchain (e.g Zephyr SDK, ...): Zephyr
 - Commit SHA or Version used: 60e8d9d8591f6d84b13501a8ff51d7a0c5f1dbbb

**Additional context**
N/A

PerMac, 2023-09-01:
I unassigned myself since I am not experience with lcov and someone else might pick it up. If not I will try looking at it

tari, 2023-10-18:
I bumped into this a few months ago while investigating support for parallel coverage data processing (`lcov --parallel`, which can speed up large reports significantly); new versions of `lcov` now treat some issues that were previously ignored as errors.

It's relatively easy to work around the issues by passing `--ignore-errors empty` and anything else it might need, but the more annoying part is detecting what options are needed for a given version of the tool: lcov 1.16 and earlier don't understand `--ignore-errors` so twister will need to probe lcov to find out what options it supports.

On the upside, supporting lcov 2.0 and future versions will also enable parallel processing with relative ease; in the Chromium embedded controller some of our reports take more than 20 minutes to process which should have something near a linear speedup when run in parallel.

PR #66406 (2023-12-15): twister: support lcov > 2.0
Support lcov > 2.0 tool which has strict error checking and some new
configuration options deprecating syntax used in 1.4 versions.

Fixes #62202
commit 7d2b3302 (2023-12-15): twister: support lcov > 2.0

Support lcov > 2.0 tool which has strict error checking and some new
configuration options deprecating syntax used in 1.4 versions.

Fixes #62202
closed 2023-12-15 by nashif