zephyrproject-rtos/zephyr #62216: Only a single, common SDU_Interval parameter in struct bt_iso_cig_param

mtpr-ot, 2023-09-01:
**Description**

ISO host interface function `bt_iso_cig_create` only allows the host application to pass a common `interval` to the controller.
The BT Core Spec v5.4 allows separate SDU_Interval to be set on C_To_P and P_To_C directions, but this is not possible with the current interface.

**Expected behavior**
There should be a `c_sdu_interval` and `p_sdu_interval`.

**Impact**
Not possible to implement the permutations for a CIS central that the EBQ Quality Suite uses. There is no error as such, but it just seems like an unnecessary limitation.

mtpr-ot, 2023-09-01:
I don't consider this a severe error, and may be explained as a conscious decision.

Thalley, 2023-09-04:
This isn't a bug, but an enhancement as the implementation works according to the design, and this would be a design change/enhancement.

closed 2023-10-23 by carlescufi